ARMmbed/mbed-os #14121: [Mbed OS 6.6.0] Flashing SDP-K1 board causes interface firmware to fail. Can no longer program board after flashing with any Mbed OS 6.6.0 project.

kylejansen, 2021-01-06:
### Description of defect
Flashing SDP-K1 with any Mbed OS 6.6.0 project appears to break the DAPLink update program preventing any more programming of the SDP-K1. Initial program will work, but any subsequent attempt to flash the board will result in a fail. Fail log is output with the following message: `error: The interface firmware FAILED to reset/halt the target MCU`.

#### Target(s) affected by this defect ?
SDP-K1

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli and online compiler

#### How is this defect reproduced ? 
Flash SDP-K1 board with any binary/project using the Mbed OS 6.6.0 library (e.g. simple LED blinky project). Program will work initially. Try to program the board with any other bin/hex file after this first flash and interface firmware will start to fail consistently.

0xc0170, 2021-01-11:
What is the previous version that worked - to start bisecting what has changed ?

cc @ARMmbed/team-st-mcd

0xc0170, 2021-01-11:
this should be though @ARMmbed/team-analogdevices

jeromecoutant, 2021-01-11:
Please check #14032

kylejansen, 2021-01-11:
> What is the previous version that worked - to start bisecting what has changed ?
> 
> cc @ARMmbed/team-st-mcd

Last working on Mbed-OS 6.5.0.

> Please check #14032

Checked this and configured the SDP-K1 so that the change in #13777 is disabled. Can confirm this is the issue, thanks Jerome. Will submit a pull request when #14032 is merged to change this option in master for the SDP-K1 to solve the problem.

jeromecoutant, 2021-01-27:
Hi
Maybe we could close this issue ?

closed 2021-01-27 by kylejansen