ARMmbed/mbed-os #14123: Example sketches for Silicon Labs Thunderboard Sense 2 doesn't compiles in the Arm MBED Online Compiler

arijitdas123student, 2021-01-07:
Hi! 👋
I was trying to run some [basic examples](https://t.co/By6bfJDYFi?amp=1) for the SiLabs Thunderboard Sense 2 using the MBED Online Compiler but I found this problem while compiling :
[image]

Can someone help me out ? 

Thanks in advance 🙂

0xc0170, 2021-01-11:
Please can you leave the original template and fill in details? Otherwise we won't be able to help you.

Also the url to an example - it should be available without logging. I'll close this , reopen with an update.

closed 2021-01-11 by 0xc0170

dhruvsheth-ai, 2021-02-19:
Hi, I'm facing the same issue. It seems that the mbed_trace.h file is missing from the provided "AMS_CCS811_gas_sensor library" provided default. Could you please update this to include the mbed_trace.h file.
Link - https://os.mbed.com/compiler/#platform:thunderboard-sense-2

Thanks
Dhruv

Telepresent, 2022-01-13:
> Hi, I'm facing the same issue. It seems that the mbed_trace.h file is missing from the provided "AMS_CCS811_gas_sensor library" provided default. Could you please update this to include the mbed_trace.h file. Link - https://os.mbed.com/compiler/#platform:thunderboard-sense-2
> 
> Thanks Dhruv

Did you manage to run the code on Thunderboard Sense 2? I need that but I also found problems. Please let me know how can I get the MBED code running.

Telepresent, 2022-01-13:
> Hi, I'm facing the same issue. It seems that the mbed_trace.h file is missing from the provided "AMS_CCS811_gas_sensor library" provided default. Could you please update this to include the mbed_trace.h file. Link - https://os.mbed.com/compiler/#platform:thunderboard-sense-2
> 
> Thanks Dhruv

I get this error: Error: Non-constant-expression cannot be narrowed from type 'int' to 'char' in initializer list [-Wc++11-narrowing] in "AMS_CCS811_gas_sensor/AMS_ENS210_temp_humid_sensor/AMS_ENS210.cpp", Line: 100, Col: 20

dhruvsheth-ai, 2022-01-13:
@Telepresent I wasn't able to solve it

0xc0170, 2022-01-13:
@stevew817 would you be able to help? Is this an issue in ams ccs811 library?

Telepresent, 2022-01-13:
> @stevew817 would you be able to help? Is this an issue in ams ccs811 library?

Yes, please. We are unable to run sample code to read sensors from Thunderboard Sense 2. I just want to read sensors and take readings out of serial port. thanks.

stevew817, 2022-01-13:
I'll see if I can find some time to look at this