ARMmbed/mbed-os #14124: [LORA] device EUI is hardcoded

jeromecoutant, 2021-01-07:
### Description of defect

In current Lora stack EUI is derived from a json configuration macro:
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/lorawan/lorastack/mac/LoRaMac.cpp#L1496

So this means that we need to update mbed_app.json file for each target 1 by 1...

This EUI should be available within an API?
or a weak function that targets can specify ?

Regards

#### Target(s) affected by this defect ?

Lora targets

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ?

40Grit, 2021-01-11:
This has been a peeve of mine for years.

40Grit, 2021-01-11:
More than 3 years actually. #6132

jeromecoutant, 2021-01-11:
@MarceloSalazar

0xc0170, 2021-01-11:
cc @ARMmbed/mbed-os-connectivity

farrenv, 2021-01-15:
This is only the case if connection parameters are not passed to the connect function.  The connect function and parameters can be executed at runtime.  If I'm missing something please elaborate.  Here is what I do in my applications at runtime:

```
    lorawan_connect_t connect_params;

        //OTAA:
    connect_params.connect_type = LORAWAN_CONNECTION_OTAA;
    connect_params.connection_u.otaa.dev_eui = DEV_EUI;
    connect_params.connection_u.otaa.app_eui = APP_EUI;
    connect_params.connection_u.otaa.app_key = APP_KEY;
    connect_params.connection_u.otaa.nb_trials = 3;

    //ABP:
    // connect_params.connect_type = LORAWAN_CONNECTION_ABP;
    // connect_params.connection_u.abp.dev_addr = DEV_ADDR;
    // connect_params.connection_u.abp.nwk_skey = NWK_SKEY;
    // connect_params.connection_u.abp.app_skey = APP_SKEY;

    retcode = lorawan.connect(connect_params);
```

farrenv, 2021-01-15:
The configuration parameters can be optionally set at runtime, so I'm not sure what the issue is?  I think I'm missing something. Please elaborate. This is what I do:
```
    lorawan_connect_t connect_params;

        //OTAA:
    connect_params.connect_type = LORAWAN_CONNECTION_OTAA;
    connect_params.connection_u.otaa.dev_eui = DEV_EUI;
    connect_params.connection_u.otaa.app_eui = APP_EUI;
    connect_params.connection_u.otaa.app_key = APP_KEY;
    connect_params.connection_u.otaa.nb_trials = 3;

    //ABP:
    // connect_params.connect_type = LORAWAN_CONNECTION_ABP;
    // connect_params.connection_u.abp.dev_addr = DEV_ADDR;
    // connect_params.connection_u.abp.nwk_skey = NWK_SKEY;
    // connect_params.connection_u.abp.app_skey = APP_SKEY;

    retcode = lorawan.connect(connect_params);
```

jeromecoutant, 2021-01-26:
Hi
Is this code from your application ?
I don't understand your point as well... you are overwriting dev_eui parameter after LoRaMac::prepare_join ?

In
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/lorawan/lorastack/mac/LoRaMac.cpp#L1496
code should be like

```
#ifdef MBED_CONF_LORA_DEVICE_EUI
        const static uint8_t dev_eui[] = MBED_CONF_LORA_DEVICE_EUI;
#else
    // get EUI from radio API
#endif
```

@40Grit ?

farrenv, 2021-02-04:
That only executes if the function is not passed params:
https://github.com/ARMmbed/mbed-os/blob/7135c6545a3b9ec7f4da094cf3ddedd6189a9469/connectivity/lorawan/lorastack/mac/LoRaMac.cpp#L1453
The code you're concerned about is all in the 'else' context.

@jeromecoutant

jeromecoutant, 2021-02-05:
OK,
So code should be updated here:
https://github.com/ARMmbed/mbed-os-example-lorawan/blob/master/main.cpp#L131

but there is no provided api to get this info for each target.

40Grit, 2021-05-13:
@jeromecoutant exactly.

This is a function that should declared as part of the mac and implemented by any BSP with a LoRa radio.

@farrenv read the thread I posted earlier in this for the original debate.