ARMmbed/mbed-os #14136: STM Nucleo-144, Pin Names in Arduino style are missing for Zio connector

sstaub, 2021-01-11:
### Description of defect

For STM Nucleo-144 boards the Arduino pin names are missing for the ST Zio connector.
The Arduino pin names are defined in PinNames.h of the target, only the classic Arduino pin names
are defined (A0-5, D0-15).
It should possible to use the Arduino style pin names of the ST Zio connector

#### Target(s) affected by this defect ?
STM32-NUCLEO-F767ZI and all other Nucleo-144 boards

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio standard toolchain

#### What version of Mbed-os are you using (tag or sha) ?

Mbed-os-6.6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.31

#### How is this defect reproduced ? 
Using a pin name of the ST Zio connector in e.g. DigitalIn gives an error
```
DigitalIn(D50); // gives an error: Use of undeclared identifier 'D50'
DigitalIn(PG_3); // must used instead
```

sstaub, 2021-01-11:
Here is a scan of the inlay from the product package. The pin numbering is also missing at the pinout of the Mbed board page.
[image]

0xc0170, 2021-01-11:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-01-11:
Hi
I agree,
but to be honest, I advice to propose a patch.
We will review and accept it ! :-)

sstaub, 2021-01-12:
@ubidefeo
I made a patch for the Nucleo-144 boards. I will attach it here as a .zip file, the paths are original.
Following PinNames.h changed:
Nucleo-F207ZG
Nucleo-F303ZE
Nucleo-F412ZG
Nucleo-F413ZH
Nucleo-F429ZI
Nucleo-F446ZE
Nucleo-F746ZG
Nucleo-F756ZG
Nucleo-F767ZI
Nucleo-H7A3ZI-Q
Nucleo-H743ZI
Nucleo-H743ZI2

[TARGET_STM.zip](https://github.com/ARMmbed/mbed-os/files/5802362/TARGET_STM.zip)

0xc0170, 2021-01-12:
Thanks @sstaub . would you be able to create a pull request?

sstaub, 2021-01-12:
I will try it.

sstaub, 2021-01-12:
PR #14144

0xc0170, 2021-01-20:
> PR #14144

Merged

closed 2021-01-20 by 0xc0170