ARMmbed/mbed-os #14139: travis: remove make.py tests (littlefs, events)

0xc0170, 2021-01-12:
### Description of defect

https://github.com/ARMmbed/mbed-os/blob/master/.travis.yml#L266 - 2 extended tests:
 - littlefs
 - events

There is quite a magic in these:

```
        # Check that example compiles
        - sed -n '/``` cpp/,/```/{/```$/Q;/```/d;p;}' ${EVENTS}/README.md > main.cpp
        - python tools/make.py -t GCC_ARM -m K64F --source=. --build=BUILD/K64F/GCC_ARM -j0
        # Check that example compiles without rtos
        - sed -n '/``` cpp/,/```/{/```$/Q;/```/d;p;}' ${EVENTS}/README.md > main.cpp
        - |
          rm -r cmsis/CMSIS_5/CMSIS/RTOS2 cmsis/device/rtos drivers/usb connectivity/cellular connectivity/drivers/cellular \
            connectivity/libraries/ppp connectivity/netsocket connectivity/nanostack connectivity/lwipstack features/frameworks/greentea-client \
            features/frameworks/utest features/frameworks/unity BUILD
        - python tools/make.py -t GCC_ARM -m NUCLEO_F103RB --source=. --build=BUILD/NUCLEO_F103RB/GCC_ARM -j0

```

This is basically events example taken from readme and compiled with baremetal - but not the same as baremetal as we have baremetal defined via config. I assume this actually differs from our baremetal profile. The tests were good to find out if we broke some legacy code. However, we should review these and possibly remove (we should already have in place the same coverage or if not, we should add it).

#### Target(s) affected by this defect ?

all

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not related

#### How is this defect reproduced ? 

Travis for each PR

0xc0170, 2021-01-12:
@hugueskamba Thanks for bringing this up.

hugueskamba, 2021-01-12:
It first builds the generated `main.cpp` with the full profile then builds it again after deleting a bunch of directories containing source code.

closed 2021-01-18 by 0xc0170