ARMmbed/mbed-os #14152: NUCLEO-F411RE debug issue

hannchen, 2021-01-14:
### Description of defect

When trying to use the debug functionality in Mbed Studio for the ST NUCLEO-F411RE, pyocd kept crashing after successfully flashing the program (e.g. mbed-os-example-blinky-baremetal). The error message read:

"0015157:ERROR:gdbserver:Unhandled exception in handle_message: STLink error (22): DP error"

Using OpenOCD or ST-LINK_gdbserver with arm-none-eabi-gdb didn't work either - the GDB server crashed as soon as I typed the continue command with a breakpoint at the main function. I then realized that single stepping worked and I found the GPIO_Full_Init function in mbed_overrides.c under targets/TARGET_STM to be the point where things start going wrong. The problem there seems to be resetting GPIOA and with that resetting the pins PA13 and PA14, which are the SWD debugging pins. After excluding those two pins from beeing reset, I am able to debug again. 

The question now is: Am I missing something here? Is there a way of telling Mbed OS not to reset those pins? Or is this really a bug?

#### Target(s) affected by this defect ?
NUCLEO-F411RE and possibly other STM32 targets (I only have the nucleo at hand).

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio build-in ARMC6 toolchain, GCC_ARM with gcc-arm-none-eabi 7.3.1 (Debian buster) or 10.2.0 (Manjaro).

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.3.1 with respective build-in tools

#### How is this defect reproduced ? 
mbed-os-example-blinky-baremetal, Target: NUCLEO-F411RE, Build profile: Debug
Try using the Mbed Studio debug functionality.

jeromecoutant, 2021-01-14:
Hi
Please try #14032

hannchen, 2021-01-14:
Hi, thanks for the fast reply. Your proposed solution works.

closed 2021-01-14 by hannchen