ARMmbed/mbed-os #14153: imgtool.py file in mbed-os interferes with imgtool python package installed in system

romanjoe, 2021-01-14:
### Description of defect

**imgtool** files included in mbed-os in **mbed-os/tools/psa/tfm/bin_utils** and particularly **imgtool.py** file interferes with **imgtool** package installed in system. 

For particular case, Cypress secure tools set called **cysecuretools** requires **imgtool** package from pip to operate. When user tries to compile a test or an example for Cypress target CY8C064B0S2_4343W he gets an error (log is provided below).

#### Target(s) affected by this defect ?

CY8C064B0S2_4343W
https://os.mbed.com/platforms/CY8CKIT-064B0S2-4343W/

#### Toolchain(s) (name and version) displaying this defect ?

gcc, arm, iar

#### What version of Mbed-os are you using (tag or sha) ?

latest master (at the time of opening https://github.com/ARMmbed/mbed-os/commit/2e905a0060bd9f0062821e55f0a0a399f1610bbd)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Name: mbed-cli
Version: 1.10.4

#### How is this defect reproduced ? 

Compile an example project for **CY8CKIT064B0S2_4343W** with **mbed compile -m CY8CKIT064B0S2_4343W -t GCC_ARM -v**

Get error:

`Traceback (most recent call last):
  File "/Users/rnok/repos/mbed-os/tools/test_api.py", line 2091, in build_test_worker
    bin_file, _ = build_project(*args, **kwargs)
  File "/Users/rnok/repos/mbed-os/tools/build_api.py", line 621, in build_project
    res = toolchain.link_program(resources, build_path, name)
  File "/Users/rnok/repos/mbed-os/tools/toolchains/mbed_toolchain.py", line 804, in link_program
    post_build_hook(self, r, elf, full_path)
  File "/Users/rnok/repos/mbed-os/tools/targets/__init__.py", line 681, in sign_image
    psoc6_sign_image(t_self, resources, elf, binf, m0hexf)
  File "/Users/rnok/repos/mbed-os/tools/targets/PSOC6.py", line 180, in sign_image
    import cysecuretools
  File "/Users/rnok/.pyenv/versions/3.7.8/lib/python3.7/site-packages/cysecuretools/__init__.py", line 16, in <module>
    from cysecuretools.main import CySecureTools
  File "/Users/rnok/.pyenv/versions/3.7.8/lib/python3.7/site-packages/cysecuretools/main.py", line 47, in <module>
    from cysecuretools.execute.signtool import SignTool
  File "/Users/rnok/.pyenv/versions/3.7.8/lib/python3.7/site-packages/cysecuretools/execute/signtool.py", line 18, in <module>
    import imgtool.main as imgtool
ModuleNotFoundError: No module named 'imgtool.main'; 'imgtool' is not a package`

0xc0170, 2021-01-19:
cc @ARMmbed/mbed-os-tools

saheerb, 2021-02-23:
@romanjoe this doesn't seem to be happening with cysecuretools=2.0.0 (this is the version we currently have in CI)

LDong-Arm, 2021-02-23:
@romanjoe Thanks for identifying the issue, that has been really helpful to us.

Here's a fix: LDong-Arm@e5901ec
We'll raise a pull request a few days later, as we are waiting for other work to be merged first, to avoid conflicts.

PR #14362 (2021-03-11): Fix imgtool import for CYTFM_064B0S2_4343W signing

### Summary of changes 

Fixes: #14153

The target CYTFM_064B0S2_4343W is signed using `cysecuretools` which depends on `imgtool` managed by `pip`. However, Arm Musca targets requires a modified/wrapped version of the image tool copied from trusted-firmware-m + MCUboot.

To avoid conflicts in the version of `imgtool` used, we should only add the copied version of `imgtool` to python's system path for Musca targets instead of doing it globally for all targets.

#### Impact of changes
commit 5ef82f59 (2021-03-11): Fix imgtool import for CYTFM_064B0S2_4343W signing

Fixes: #14153

The target CYTFM_064B0S2_4343W is signed using `cysecuretools` which
depends on `imgtool` managed by `pip`. However, Arm Musca targets
requires a modified/wrapped version of the image tool copied from
trusted-firmware-m + MCUboot.

To avoid conflicts in the version of `imgtool` used, we should only
add the copied version of `imgtool` to python's system path for
Musca targets instead of doing it globally for all targets.
closed 2021-03-11 by 0xc0170