ARMmbed/mbed-os #14171: Error when using chrono on function

felipeLeast, 2021-01-19:
### Description of defect
We've first notice this error when we updated our code to chrono to avoid warnings but then some functionalities stop working, like our delay to send packets for example, and we've noticed that it occours because chrono milliseconds function can't be used on function call, because when we've changed it to:

```c
auto wait_for_ms_chrono = std::chrono::milliseconds(wait_for_ms);
queue.call_in(wait_for_ms_chrono, call_queue, arg);
```

it start working again, we've also seen this behaviour inside one of the mbed driver, the NanostackRfPhy2slp, when using `S2LP_USE_CHRONO` macro the cca timer attach uses `microsecond` and the cca_timer begins to act not acording to slots timeout like it's supposed to. If our IC is a little bit slower it will stop receiving ACK because the CCA timer won't wait long enough to receive it.

#### Target(s) affected by this defect ?
NUCLEO_WB55RG, NRF52_DK

#### Toolchain(s) (name and version) displaying this defect ?
arm gcc 9.3.1 20200408

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.4

#### How is this defect reproduced ? 

Create an event queue and call for attach with `microsencods` or `milliseconds` from chrono and check if timeout will be the same as requested.

0xc0170, 2021-01-20:
> Create an event queue and call for attach with microsencods or milliseconds from chrono and check if timeout will be the same as requested.

@felipeLeast could you create a code snippet to reproduce a problem - what works and what does not?

@kjbracey-arm if you spot any issue meanwhile

kjbracey-arm, 2021-01-21:
There shouldn't be much scope for failure here, as the deprecated calls are just based on the Chrono stuff anyway:

    template <typename F, typename... ArgTs>
    MBED_DEPRECATED_SINCE("mbed-os-6.0.0", "Pass a chrono duration, not an integer millisecond count. For example use `5s` rather than `5000`.")
    int call_in(int ms, F f, ArgTs... args)
    {
        return call_in(duration(ms), std::move(f), args...);
    }

Hang on, if I read correctly, you've posted an example of what *does* work? You need to post what *doesn't* work.

I'll have a look at that driver you mention...