ARMmbed/mbed-os #14174: "target.header_offset" and "target.app_offset" ignored when defined in mbed_lib.json

TaniaMirzin, 2021-01-19:
Hi, am working with mbed-os https://github.com/ARMmbed/mbed-os/#f2278567d09b9ae9f4843e1d9d393526b9462783
and I am using a prebuild bootloader. As I am using bootloader not  from mbed-os\features\FEATURE_BOOTLOADER\, the "BOOTLOADER" feature is turned off in my mbed_app.json file. 
My prebuild bootloader directory has its own mbed_lib.json file with path to the specific bootloader
bin file, header format , "target.app_offset" and "target.header_offset" configurations.
It seems that all values are taken from this mbed_lib.json files except "target.app_offset"
and "target.header_offset" values. It is seems that when a build occurs, the
application placed immediately after the bootloader and ignoring these values
of header and application offset. Is this behavior ok? What should I make to
take these configs from my mbed_lib.json?

- The issue can be reproduced with https://github.com/PelionIoT/mbed-cloud-client-example-internal/tree/mbed-os_issue.
Used mbed_app_nextgen.json file.

- mbed cli --version  1.10.2

- mbed-os https://github.com/ARMmbed/mbed-os/#f2278567d09b9ae9f4843e1d9d393526b9462783

- DISCO_L475VG_IOT01A target

output   : 
```
Building project mbed-cloud-client-example-internal (DISCO_L475VG_IOT01A, GCC_ARM)
Scan: mbed-cloud-client-example-internal
Using ROM regions bootloader, header, application, post_application in this build.
  Region bootloader: size 0x6800, offset 0x8000000
  Region header: size 0x50, offset 0x8008000
  Region application: size 0x77600, offset 0x8008200
  Region post_application: size 0x80800, offset 0x807f800
```
When header and application should be taken from 
prebuilt-bl\TARGET_DISCO_L475VG_IOT01A\TARGET_BL_INTERNAL_FLASH\mbed_lib.json
with these configurations:
```
{
    "name": "mbed-bootloader-DISCO_L475VG_IOT01A-TARGET_BL_INTERNAL_FLASH",
    "target_overrides": {
        "*": {
            "target.macros_add": [
                "MBED_BOOTLOADER_FLASH_BANK_SIZE=MBED_ROM_SIZE/2",
                "MBED_CONF_STORAGE_STORAGE_TYPE=TDB_INTERNAL",
                "MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_BASE_ADDRESS=(MBED_ROM_START + MBED_BOOTLOADER_FLASH_BANK_SIZE)",
                "MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_SIZE=(2*16*1024)",
                "MBED_CLOUD_CLIENT_FOTA_BLOCK_DEVICE_TYPE=FOTA_INTERNAL_FLASH_BD",
                "MBED_CLOUD_CLIENT_FOTA_STORAGE_START_ADDR=(MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_BASE_ADDRESS + MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_SIZE)",
                "MBED_CLOUD_CLIENT_FOTA_STORAGE_SIZE=(MBED_BOOTLOADER_FLASH_BANK_SIZE - MBED_CONF_STORAGE_TDB_INTERNAL_INTERNAL_SIZE)"

            ],
            "target.bootloader_img": "mbed-bootloader-e7ebbc1.bin",
            "target.header_offset" : "0x8000",
            "target.app_offset"    : "0x8200",
            "target.restrict_size" : "0x77E00",
            "target.header_format": [
                ["magic", "const", "32le", "0x5c0253a3"],
                ["firmwareSize", "size", "32le", ["application"]],
                ["firmwareVersion", "const", "64le", "0x0"],
                ["firmwareDigest", "digest", "SHA256", "application"],
                ["reserved", "digest", "SHA256", "application"]
            ]
        }
    }
}
```

0xc0170, 2021-01-19:
According to the docs, https://os.mbed.com/docs/mbed-os/v6.6/program-setup/bootloader-configuration.html , it should work.

@ARMmbed/mbed-os-tools please review

jrobeson, 2021-01-21:
you are using the mbed cli and not the beta https://github.com/ARMmbed/mbed-tools for this right? It works for me with the mbed cli, but does not yet work with mbed-tools

TaniaMirzin, 2021-01-21:
@jrobeson I am using mbed-cli --version 1.10.2

ThomasGonet, 2021-07-26:
Hello, same problem here with mbed-cli version 1.10.5. 

Only workaround is to put this params into mbed-app.json for the moment.