ARMmbed/mbed-os #14182: Mbed breaks the debugger interface on the Disco-L475VG-IOT01A

ithinuel, 2021-01-20:
### Description of defect

Mbed does something to the debug interface that prevents common debugger to connect normally to board.
Only `connect under reset` works, and the debugger session only works for the early stages of boot. As Mbed/sdk kicks in the connection to the target is lost.

#### Target(s) affected by this defect ?

`disco l475vg iot01a`

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

2e905a0060b

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

openocd 0.10.0
pyocd 0.28.3
mbed 1.10.5

#### How is this defect reproduced ? 

build a blinky or a test app.
Try to step through it using gdb connected to openocd or pyocd.

jeromecoutant, 2021-01-20:
Do you have #14032 ?

ithinuel, 2021-01-21:
I did not ! It seems to work fine now.
Thank you !

closed 2021-01-21 by ithinuel