ARMmbed/mbed-os #14194: Unable to correctly stop BLE advertising

RobVlaar, 2021-01-26:
### Description

 - type: bug

---
### Bug

**Target**
NRF52840

**Files**
gapImpl.cpp

**mbed-os**
6.4.0

**Issue**
StopAdvertising always returns `BLE_STACK_BUSY`.
When I start advertising, only the `_pending_sets` gets set.
The `_active_sets` doesn't get set.
but in the `stopAdvertising` the function always returns `BLE_STACK_BUSY` because of : 
```
if (!_active_sets.get(handle) || _pending_sets.get(handle)) {
        return BLE_STACK_BUSY;
    }
``` 
Currently I use the reset function to stop the advertising.

**Question**
Is the if statement incorrect?

When I remove the if statement ble Advertising is stopped and then it is possible to shutdown down the ble to be able to go low power ( with some other modifications to NRFCordioHCIDriver shutdown to stop timer 0 and timer 2)

pan-, 2021-01-26:
Thanks for reporting the issue @RobVlaar Could you detailed what is done on the user side of things ? 
The advertising set is supposed to be set to pending until the stack reports that is has been properly enabled. That event is reported to the application in `Gap::EventHandler::onAdvertisingStart` ([here](https://github.com/ARMmbed/mbed-os/blob/0c6753bb821612eb72b5f682aed7ffcc98e80a1f/connectivity/FEATURE_BLE/include/ble/Gap.h#L324)).

RobVlaar, 2021-01-27:
Thanks for the awnser @pan- . hmm, we don't use the onAdvertisingStart. We only usethe onConnectionComplete and onDisconnectionComplete. Maybe we just use the functions wrong. 

how we implemented the BLE in short: we set the gap eventhandler and init BLE,  when the init is completed we configure the advertising data. After that we start advertising and start our own ble timeout.
When ble is connected via onConnectionComplete we only set a led to blue. After that we do nothing untill we get a disconnect via onDisconnectionComplete or when the timeout has passed we do a shutdown. When we get a onDisconnectComplete we do a shutdown. 
In the shutdown we try to stop the advertising ( which fails ) and shutdown the BLE.
Thats basicly it. 

Everything is functional but only the stop advertising doesn't work.

pan-, 2021-01-27:
@RobVlaar I suggest you override the `onAdvertisingStart` and `onAdvertisingStop` callbacks and add a trace inside to understand when advertising start and when advertising end. 

From what I understand you connect to the target so advertising has already stopped!
 
Not all Bluetooth CHIP supports advertising to continue when they are connected. We made the choice in mbed BLE API to stop (non extended) advertising when the advertiser gets connected. The goal was to avoid different behavior depending in the Bluetooth controller used. 

However I think it would be fair to not return `BLE_STACK_BUSY` if the advertising set is not active. 

@paul-szczepanek-arm Could you look into it ?

paul-szczepanek-arm, 2021-01-27:
https://github.com/ARMmbed/mbed-os/pull/14202

paul-szczepanek-arm, 2021-02-03:
merged

paul-szczepanek-arm, 2021-02-05:
@RobVlaar does this solve your problem?

closed 2021-02-10 by ciarmcom