ARMmbed/mbed-os #14208: STM32 HAL_GetTick can calculate wrong elapsed time

Uruloke, 2021-01-28:
### Description of defect

The HAL_GetTick override is called by the HAL drivers and mbedtls on STM platforms. It is used for time-keeping and timeout management. The HAL_GetTick function calculates the elapsed time between calls using the us_ticker by looking at the difference between the previous read value and the current time of us_ticker. The previous timestamp is stored in a global variable until next time it's used.

The current implementation of HAL_GetTick is this:

```
uint32_t HAL_GetTick()
{
    uint32_t new_time = us_ticker_read();
    uint32_t elapsed_time = (((new_time - prev_time) & US_TICKER_MASK) + prev_tick_remainder);
    prev_time = new_time;
    uint32_t elapsed_ticks;
    // Speed optimisation for small time intervals, avoiding a potentially-slow C library divide.
    if (TIM_MST_BIT_WIDTH <= 16 || elapsed_time < 65536) {
        elapsed_ticks = 0;
        while (elapsed_time >= 1000) {
            elapsed_ticks++;
            elapsed_time -= 1000;
        }
        prev_tick_remainder = elapsed_time;
    } else {
        elapsed_ticks = elapsed_time / 1000;
        prev_tick_remainder = elapsed_time % 1000;
    }
    total_ticks += elapsed_ticks;
    return total_ticks;
}
```

A situation can occur where HAL_GetTick calculates the wrong elapsed time if a thread switch happens at the wrong time. The HAL_GetTick is not protected against thread switching at all so if multiple threads in the system makes calls to the HAL_GetTick function then a thread can take control while another is executing HAL_GetTick after new_time is read and before prev_time is used and update the prev_time global variable. When the original thread can continue execution it will use the newly updated prev_time (which will now be later in time than new_time). This will cause HAL_GetTick to think that a wrap has occurred on the elapsed time and subsequently anyone using the time from HAL_GetTick will think more time has elapsed than it actually has.

I think the only solution here is to make HAL_GetTick run in a critical section, disabling interrupts until it is done. This will prevent thread switching to happen and ensure that elapsed time is always calculated correctly.

In trying to track this down I ran the system using instruction tracing and the following screenshots show this exact scenario happen.

This is an overview of the timeline where the issue happens. First call to HAL_GetTick is at 1 quickly followed by a thread switch. Another thread calls HAL_GetTick in 2. Finally the original caller gets to finish it's execution of HAL_GetTick at 3.
[image]

Original call to HAL_GetTick which gets thread switched:
[image]

Another thread calls HAL_GetTick and updates prev_time:
[image]

Original call to HAL_GetTick gets to continue, now using the value of prev_time set by the previous thread:
[image]

#### Target(s) affected by this defect ?

All STM targets using HAL_GetTick().

#### Toolchain(s) (name and version) displaying this defect ?

gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (GNU Tools for Arm Embedded Processors 9-2019-q4-major)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

1. Have multiple threads make use of HAL APIs that use HAL_GetTick.
2. Eventually HAL_GetTick will return a much larger value than have actually elapsed. Timing will be random due to the when calls happen in relation to SysTick thread switching.

Uruloke, 2021-01-28:
I've tested this on version 6.3.0 but nothing have changed in hal_tick_overrides.c since then and so this is also an issue with the latest master branch.

0xc0170, 2021-01-28:
cc @ARMmbed/team-st-mcd

flippy84, 2021-01-29:
Submitted a pull request that fixes this issue
#14215

0xc0170, 2021-01-29:
Thanks @flippy84 , we will review

jeromecoutant, 2021-02-03:
#14215 is merged!

jeromecoutant, 2021-02-23:
@0xc0170 Let's close ?
Thx

closed 2021-02-23 by 0xc0170