ARMmbed/mbed-os #14220: CMake: Build multiple binaries from single mbed-os source

nat45928, 2021-02-01:
### Description of defect

Follow up from #12837 

I am trying to create two binaries from a single repo using the new CMake build system but not having much sucess. I have setup a root CMakeLists.txt which creates a base library target which depends on the `mbed-os` build sources and adds some common project sources. In two folders I have a CMakeLists.txt which adds an executeable and additional target specific source files.

When building I either get multiple copies of mbed-os (instead of a single build follow by linking for each executable) or a get errors about multiple definitions / symbols.

I'm fairly new to CMake-based projects so this might just be user error, but I'm not sure what the right definitions are.

The project structure looks like the following:

```
root
 |-> mbed-os
 |-> common
 |   |-> include
 |   |-> src
 |-> foo
 |   |-> CMakeLists.txt
 |   |-> main.cpp
 |-> bar
 |   |-> CMakeLists.txt
 |   |-> main.cpp
 |-> CMakeLists.txt
 |-> mbed-lib.json
```

The root CMakeLists.txt

```cmake
cmake_minimum_required(VERSION 3.19.0 FATAL_ERROR)

set(MBED_PATH ${CMAKE_CURRENT_SOURCE_DIR}/mbed-os CACHE INTERNAL "")
set(MBED_CONFIG_PATH ${CMAKE_CURRENT_SOURCE_DIR}/.mbedbuild CACHE INTERNAL "")

include(${MBED_PATH}/tools/cmake/app.cmake)
add_subdirectory(${MBED_PATH})

add_library(project-common)

project(my-project)

target_sources(
    project-common
    PRIVATE
        common/src/utility.cpp
)
target_include_directories(
    project-common
    PUBLIC
        .
        common/include
)

target_link_libraries(
    project-common
    PUBLIC
        mbed-os
        mbed-events
)

# add binary directories
add_subdirectory(foo)
add_subdirectory(bar)
```

And the `foo/CMakeLists.txt` file looks like:

```cmake
add_executable(foo)

mbed_configure_app_target(foo)
mbed_set_mbed_target_linker_script(foo)

target_sources(
    foo
    PRIVATE
        main.cpp
)

target_link_libraries(
    foo
    PRIVATE
        project-common
)

mbed_set_post_build(foo)
```

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.7.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools: 5.3.0

#### How is this defect reproduced ?

nat45928, 2021-02-01:
Added some extra details.

ladislas, 2021-02-01:
@nat45928 I'm in the same situation as you, please take a look here: https://github.com/ARMmbed/mbed-os/issues/13981

0xc0170, 2021-02-01:
Thanks @nat45928 for reporting this issue. I'll use your details to reproduce it locally with a simple demo, and lets see how we can fix it (one of the solution is provided in #13981)

multiplemonomials, 2021-10-03:
This appears to have been fixed by #14275, adding multiple apps works fine now.