ARMmbed/mbed-os #14222: USBSerial not working in combination with CAN on STM32 bluepill

mennodegraaf, 2021-02-01:
### Description of defect

I am using the STM32 bluepill as a custom target with USB enabled. USBSerial is then working (pins PA_11, PA_12), however if I also use CAN (on PB_8, PB_9) there is no USBSerial device being detected anymore.

#### Target(s) affected by this defect ?

STM32 bluepill STM32F103C8

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6 from Mbed Studio 1.3.1 Linux

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Simple example program:

`#include "mbed.h"
#include "USBSerial.h"

DigitalOut led1(LED1);      // Onboard LED
USBSerial serial(false);    // Non-blocking
CAN can(PB_8, PB_9);    // Enabling onboard CAN breaks USBSerial
Ticker t;                   // Ticker with microsecond resolution

// Redirect console output to usb-serial port
FileHandle *mbed::mbed_override_console(int fd)
{
    return &serial;
}

// Periodically called
void onTick(void) {
    led1 = !led1;
}

int main()
{
    t.attach(&onTick,200ms);      // periodically call the onTick function

    printf("This is a USB serial example on Mbed OS %d.%d.%d.\n", MBED_MAJOR_VERSION, MBED_MINOR_VERSION, MBED_PATCH_VERSION);

    while (true)
    {
        printf("USBSerial test\n");
        ThisThread::sleep_for(1s);
    }
}`

0xc0170, 2021-02-02:
Have you looked at it closer what CAN actually configured in the HAL? Does it reconfigure anything that USB needs or is there a pin conflict or something else there going on?

cc @ARMmbed/team-st-mcd

mennodegraaf, 2021-02-02:
Well, the pins for onboard USB (PA_11, PA_12) are an alternative configuration for CAN. In the above example however, CAN is configured on pins PB_8, PB_9. I agree that this alternative configuration may well be the source of the issue. I am still relatively new to mbed-os, so please guide me in the right direction as to where I have to look for what is happening under the hood when CAN is used.

jeromecoutant, 2021-02-23:
Hi

I don't see yet any relation between CAN (using PB8/9) and USB (using PA11/12)...

> mbed-os-6.6.0

A bug issue was  included in this version... Maybe you should try the latest one ?

Thx

mennodegraaf, 2021-02-24:
Updated mbed-os to 6.8.0, exactly the same issue. At the moment, I uncomment CAN in the above example, the USB serial stops working.

jeromecoutant, 2021-02-24:
Hi
Issue is reproduced with NUCLEO_F103RB... :-(

derneue, 2021-04-06:
This is not a bug! See the note in the RM0008 page 654:

In low, medium-, high- and XL-density devices the USB and CAN share a dedicated 512-byte SRAM memory for data transmission and reception, and so they cannot be used concurrently (the shared SRAM is accessed through CAN and USB exclusively). The USB and CAN can be used in the same application but not at the same time.

Please close this issue.

mennodegraaf, 2021-04-06:
OK, so this is a functional limitation of the STM32F1 series. Unfortunately, it was not part of the datasheet, but could only be found in the reference manual. Anyway, I wonder if some error message in Mbed-OS is useful to indicate that this combination is not valid?

jeromecoutant, 2021-04-08:
You could imagine some global state updated and/or checked in USB and CAN init function...

or a simple documentation update in
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/README.md#stm32-families

I let you proposing the patch, we will review it.
Thx

MarceloSalazar, 2021-04-28:
Closing as requested.

closed 2021-04-28 by MarceloSalazar