ARMmbed/mbed-os #14226: "Error: Use of undeclared identifier 'wait' in" usage of deprecated functions

burcus, 2021-02-02:
### Description of defect

I'm compiling a program with online mbed-os compiler. When i try to compile my program there are some errors like:
Error: Use of undeclared identifier 'wait' in .....
Error: Use of undeclared identifier 'wait_ms' in ....
Error: Use of undeclared identifier 'Serial' in

Before describing my problem, i want to specify that I viewed similar issues and solutions.
There is different suggestions like "wait_ns, wait_us, ThisThread::sleep_for
BufferedSerial for serial. 

The issues I don't understand are:
wait() function gets "seconds" parameter as "float"
wait_ms() function gets "milliseconds" parameter as "int"

If I understood correctly ThisThread::sleep_for() gets ms parameter as "int" 

When I use ThisThread::sleep_for instead of wait_ms there is no problem. wait_ms(2000) will be ThisThread::sleep_for(2000)
But how can i use it for wait()? My parameter is float seconds like "0.02" ThisThread::sleep_for() gets int. 

Also is there a way compile it without any changing in functions?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

I'm using online mbed compiler version 1.10.25.0

0xc0170, 2021-02-02:
Please leave a default issue template with information. Plus what code are you compiling? do you include `mbed.h` in your app?

burcus, 2021-02-02:
Hi there,

I edited it. I'm using online mbed ide. I don't have too many experience with this. I hope I answered correctly to questions in template. Yes I have mbed.h in my codes. The project that I'm working on is a developed project by others and I'm changing or adding some functionalities it. Actually codes were compiling, it seems to mbed 2. I didn't update it to mbed os 6 but codes effected from mbed os 6 changes.

My codes are parts of project files. But for example: when i'm updating my functions according to mbed-os 6 documentations. My driver files throws error e.g wait() functions in the SX1272_LoRaRadio.cpp .

0xc0170, 2021-02-02:
That explains it. the functions are not available in Mbed OS 6 version, they were deprecated long time ago and removed in 6.x. You will need to update your codebase to use functions like you already mentioned `sleep_for`.

`BufferedSerial ` has set_baud methond, not `baud`. Look at the declarations in the header files  or documentation that should have these function mentioned (equally functionality).

I close this as it is not a technical bug at the moment.

closed 2021-02-02 by 0xc0170