ARMmbed/mbed-os #14231: CY8CKIT064B0S2_4343W: Missing policy file for post build operation

hugueskamba, 2021-02-03:
### Description of defect
The `CY8CKIT064B0S2_4343W ` Mbed target does not successfully build as the post build operation cannot complete. Signing the image [requires a policy file (`policy_single_CM0_CM4.json`)](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L7034) which is currently missing from the repository.

#### Target(s) affected by this defect ?
`CY8CKIT064B0S2_4343W`

#### Toolchain(s) (name and version) displaying this defect ?
`ARM` or `GCC_ARM`

#### What version of Mbed-os are you using (tag or sha) ?
`mbed-os-6.7.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
```
$ mbed-cli --version
1.10.5
```

#### How is this defect reproduced ? 
Run the following command:
`mbed compile -t ARM -m CY8CKIT064B0S2_4343W`

Result:
```
...
Compile [100.0%]: cy_usbfs_dev_drv_io_dma.c
Link: mbed-os-example-blinky
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
Elf2Bin: mbed-os-example-blinky
Post-Build: mbed-os-example-blinky
M0 core image file found: ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CKIT064B0S2_4343W/device/COMPONENT_CM4/hex/psoc6_02_cm0p_secure.hex.
Policy file /path/to/mbed-os-example-blinky/policy_single_CM0_CM4.json not found. Aborting.
[ERROR] Required policy file not found.
[mbed] ERROR: "/usr/local/opt/python@3.9/bin/python3.9" returned error.
       Code: 1
       Path: "/path/to/mbed-os-example-blinky"
       Command: "/usr/local/opt/python@3.9/bin/python3.9 -u /path/to/mbed-os-example-blinky/mbed-os/tools/make.py -t ARM -m CY8CKIT064B0S2_4343W --source . --build ./BUILD/CY8CKIT064B0S2_4343W/ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

0xc0170, 2021-02-03:
cc @ARMmbed/team-cypress

Patater, 2021-02-22:
We are currently implementing new CI for testing CMake-based builds. I recall we had to do something in CI to make this file present.

@jamesbeyond How did we get this target building in our other CI before?

jamesbeyond, 2021-02-22:
> We are currently implementing new CI for testing CMake-based builds. I recall we had to do something in CI to make this file present.
> 
> @jamesbeyond How did we get this target building in our other CI before?

Build tool complains file is missing, it is because unlike other mbed targets,  some cypress targets will require some extra steps to generate the policy and key files before it start builds, the required steps are:
- install cysecuretools>=2.1.0  https://pypi.org/project/cysecuretools
- cd targets/TARGET_Cypress/TARGET_PSOC6/TARGET_${targetName}
- cysecuretools -t cy8ckit-064b0s2-4343w init
- cysecuretools -t cy8ckit-064b0s2-4343w create-keys

The required targets are:
- CYTFM_064B0S2_4343W
- CY8CKIT064B0S2_4343W

Patater, 2021-02-23:
I see two options that'd help this UX:
1. Run these commands automatically from CMake if the files are missing, printing a message explaining both why and what we are doing
2. If the files are missing when building for these targets, display an error message describing the commands to run in order to get the build working

hugueskamba, 2021-02-23:
@jamesbeyond 
I did:
1. `cd targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CKIT064B0S2_4343W/`
1. `cysecuretools -t cy8ckit-064b0s2-4343w init`
1. `cysecuretools -t cy8ckit-064b0s2-4343w create-keys`

Here are the policy files generated:
```
hugkam01 at C02WD1BEHV2R in /path/to/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CY8CKIT064B0S2_4343W on master [!]
$ l policy/
total 64
drwxr-xr-x   6 hugkam01  1399503260   192B 23 Feb 18:10 .
drwxr-xr-x  15 hugkam01  1399503260   480B 23 Feb 18:11 ..
-rw-r--r--   1 hugkam01  1399503260   5.4K 23 Feb 18:10 policy_multi_CM0_CM4_smif_swap.json
-rw-r--r--   1 hugkam01  1399503260   5.4K 23 Feb 18:10 policy_multi_CM0_CM4_swap.json
-rw-r--r--   1 hugkam01  1399503260   5.2K 23 Feb 18:10 policy_single_CM0_CM4_smif_swap.json
-rw-r--r--   1 hugkam01  1399503260   5.2K 23 Feb 18:10 policy_single_CM0_CM4_swap.json
```

The file `policy_single_CM0_CM4.json` is still missing

ifyall, 2021-02-23:
@hugueskamba, what version of cysecuretools are you running when using those commands?
Thanks,
Ian

RaymondNgun, 2021-02-23:
@hugueskamba, please install cysecuretools 2.1.0 e.g. pip install cysecuretools==2.1.0

hugueskamba, 2021-02-23:
> @hugueskamba, what version of cysecuretools are you running when using those commands?
> Thanks,
> Ian

Hi,
This is the version I am currently running:
```
$ cysecuretools version

Package:
	CySecureTools: 3.0.0
	CyBootloader: 
		PSoC64 1M: 2.0.0.3257
		PSoC64 2M: 2.0.0.3345
		PSoC64 512K: 2.0.0.3257
```

hugueskamba, 2021-02-23:
> @hugueskamba, please install cysecuretools 2.1.0 e.g. pip install cysecuretools==2.1.0

With version 2.1.0

I get the following files:
```
$ l policy/
-rw-r--r--   1 hugkam01  1399503260   4.9K 23 Feb 21:11 policy_multi_CM0_CM4.json
-rw-r--r--   1 hugkam01  1399503260   4.9K 23 Feb 21:11 policy_multi_CM0_CM4_smif.json
-rw-r--r--   1 hugkam01  1399503260   4.7K 23 Feb 21:11 policy_single_CM0_CM4.json
-rw-r--r--   1 hugkam01  1399503260   4.7K 23 Feb 21:11 policy_single_CM0_CM4_smif.json
```

Thanks

hugueskamba, 2021-02-23:
@jamesbeyond will you be adding the extra steps in the CI to fix the CMake test build failure?

hugueskamba, 2021-03-25:
The CI is now configuring the environment correctly to build artefacts for the board.

closed 2021-03-25 by hugueskamba