ARMmbed/mbed-os #14234: Heap stats do not change after malloc() on NRF52_DK

OllyMcBrideOB, 2021-02-03:
### Description of defect

When running the example code & configuration from [Heap Statistics](https://os.mbed.com/docs/mbed-os/v5.12/tutorials/optimizing.html) tutorial, the heap stats do not change after a successful malloc(). 

#### Target(s) affected by this defect ?

NRF52_DK

#### Toolchain(s) (name and version) displaying this defect ?
n/a

#### What version of Mbed-os are you using (tag or sha) ?

[d9819a72d1d2daea0609e921efde49e9ba06c9b7](https://github.com/ARMmbed/mbed-os/commit/ba6f566bd8c97973de24cdc57f087b56a7bb8f72)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 

Using the instructions on the [Heap Statistics](https://os.mbed.com/docs/mbed-os/v5.12/tutorials/optimizing.html) tutorial, I have copied and pasted `main()` and the `mbed_app.json` configuration into `mbed-os-example-blinky` and tried to run it on a Nordic NRF52832 Dev Kit (PCA10400). 

The output of the serial monitor is shown below:

```
Starting heap stats example
Start; Current heap: 64
Start; Max heap size: 64

Allocating 1000 bytes
Post-Alloc; Current heap: 64
Post-Alloc; Max heap size: 64

Freed 1000 bytes
Post-Free; Current heap: 64
Post-Free; Max heap size: 64
```

However, there is no change in the 'Post-Alloc' heap size. I would expect to see the following to indicate that the `MBED_HEAP_STATS_ENABLED` is enabled and the `malloc()` was successful:
```
Starting heap stats example
Start; Current heap: 64
Start; Max heap size: 64

Allocating 1000 bytes
Post-Alloc; Current heap: 1064
Post-Alloc; Max heap size: 1064

Freed 1000 bytes
Post-Free; Current heap: 64
Post-Free; Max heap size: 1064
```
Or I would expect to see the following, which should indicate that `MBED_HEAP_STATS_ENABLED` is not enabled:
```
Starting heap stats example
Start; Current heap: 0
Start; Max heap size: 0

Allocating 1000 bytes
Post-Alloc; Current heap: 0
Post-Alloc; Max heap size: 0

Freed 1000 bytes
Post-Free; Current heap: 0
Post-Free; Max heap size: 0
```

My initial thought was that the `MBED_HEAP_STATS_ENABLED` configuration was not being set properly, resulting in the stdlib `malloc()` being called instead of the mbed `malloc_wrapper()`, which includes the heap stats tracing. 

The output from `mbed compile --config -tARM -mNRF52_DK` seems to indicate that the configuration setting is set correctly (see image).

[image]

I'm currently out of ideas as to why the heap stats doesn't function as expected, 

For reference, the image below shows a screenshot of the program, configuration and unexpected output.

[image]

pan-, 2021-02-09:
That's a great description. Thanks @OllyMcBrideOB . Have you tried with a different compiler like GCC ? 

I tried with GCC on an NRF52840 and I obtain the following result: 

```
Starting heap stats example
Start; Current heap: 1034
Start; Max heap size: 1034

Allocating 1000 bytes
allocation = 0x20002fd0
Post-Alloc; Current heap: 2034
Post-Alloc; Max heap size: 2034

Freed 1000 bytes
Post-Free; Current heap: 1034
Post-Free; Max heap size: 2034
```

--- 

@evedon It seems that the doc promote the wrong method to enable the heap statistics, it should be 

```json
    "target_overrides": {
        "*": {
            "platform.heap-stats-enabled": true
        }
    }
```

and not 

```json
{
    "macros": [
        "MBED_HEAP_STATS_ENABLED=1"
    ],
    ...
}
```

evedon, 2021-02-10:
@pan- You are right that we should update the tutorial to promote using `platform.heap-stats-enabled` and not the macro name. The documentation page on the [configuration system](https://os.mbed.com/docs/mbed-os/v6.7/program-setup/advanced-configuration.html#using-configuration-data-in-code) correctly discourages the usage of the macro.

OllyMcBrideOB, 2021-02-11:
Hi @pan-,

I have changed my `mbed_app.json` file to use:
```
{
"target_overrides": {
        "*": {
            "platform.heap-stats-enabled": true
        }
    }
}
```

And have compiled for NRF52_DK using GCC_ARM (running on a NRF52832 PCA10400 dev kit) and I get the following:

```
Starting heap stats example
Start; Current heap: 1024
Start; Max heap size: 1024

Allocating 1000 bytes
Post-Alloc; Current heap: 1024
Post-Alloc; Max heap size: 1024

Freed 1000 bytes
Post-Free; Current heap: 1024
Post-Free; Max heap size: 1024
```

When I run `mbed compile -m NRF52_DK -t GCC_ARM --config` I have confirmed that the heap-stats-enabled configuration is set correctly:
```
platform.heap-stats-enabled = 1 (macro name: "MBED_HEAP_STATS_ENABLED")
```

For reference, the image below shows a screenshot of the program now compiled using GCC_ARM, the changed `mbed_app.json` configuration and unexpected output.

[image]

0xc0170, 2021-02-15:
enable also `MBED_MEM_TRACING_ENABLED`, see https://os.mbed.com/blog/entry/Tracking-memory-usage-with-Mbed-OS/ for details.

OllyMcBrideOB, 2021-02-18:
I've just revisited this issue and have managed to get the Heap Stats working on the NRF52832 using the GCC_ARM compiler. 

Using Mbed Studio, it turns out I didn't properly configure it to use GCC_ARM, and therefore it was still using ARM6. As soon as I followed the instructions [here](https://os.mbed.com/docs/mbed-studio/current/installing/switching-to-gcc.html) and configured Mbed Studio for GCC_ARM, the Heap Stats functionality is working successfully. 

Is it possible to either enable this functionality with the ARM6 compiler? Or to add a note to the [Heap Statistics](https://os.mbed.com/docs/mbed-os/v5.12/tutorials/optimizing.html) tutorial to specify that it will only work with GCC_ARM?