ARMmbed/mbed-os #14236: Correct linker option for NUCLEO_WL55JC

JeanMarcR, 2021-02-04:
### Description of defect
To be able to run correctly the blink application on the target NUCLEO_WL55JC, I need to add "--cpu=Cortex-M4.no_fp" in the linker’s file : develop.cmake.
Despite the fact that “mcpu=cortex-m4 and -mfpu=none” is present in the compiler’s file : Cortex-M4.cmake
The blink application is obtained thanks to the following command:
mbedtools compile -t ARM -m NUCLEO_WL55JC

#### Target(s) affected by this defect ?
NUCLEO_WL55JC

#### Toolchain(s) (name and version) displaying this defect ?
mbedtools 4.6.1

#### What version of Mbed-os are you using (tag or sha) ?
6dd44e7ec77d2e3c488022af8f4dc310c141aaf5
 ("Merge pull request #14190 from 0xc0170/fix_ns_cores", 2021-02-03)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
-- The C compiler identification is ARMClang 6.13.100
-- The CXX compiler identification is ARMClang 6.13.100
-- The ASM compiler identification is ARMClang
-- Found assembler: C:/Keil_v5/ARM/ARMCLANG/bin/armclang.exe

#### How is this defect reproduced ? 

If I removed "--cpu=Cortex-M4.no_fp" of the linker’s file : develop.cmake
the blink application does not run correctly on the target.

0xc0170, 2021-02-05:
Thanks for the report, we will look at it.

cc @hugueskamba

hugueskamba, 2021-02-05:
Hi @JeanMarcR 
As you have the board which is failing to run Blinky, I suggest you add `list(APPEND link_options "--cpu=Cortex-M4.no_fp")` in `tools/cmake/cores/Cortex-M4.cmake`. If it solves your issue please let us know so a PR fixing this can be raised.
Thanks

0xc0170, 2021-02-05:
how come we do not see it with the other targets?

jeromecoutant, 2021-02-05:
> how come we do not see it with the other targets?

This is the first "Cortex-M4" target

hugueskamba, 2021-02-16:
> Hi @JeanMarcR
> As you have the board which is failing to run Blinky, I suggest you add `list(APPEND link_options "--cpu=Cortex-M4.no_fp")` in `tools/cmake/cores/Cortex-M4.cmake`. If it solves your issue please let us know so a PR fixing this can be raised.
> Thanks

@JeanMarcR Have you tried the suggestion above?

JeanMarcR, 2021-02-18:
Hi
Yes.
A PR has been pushed.
https://github.com/ARMmbed/mbed-os/pull/14183.

I suggest to close the issue ?

hugueskamba, 2021-02-18:
Fixed with https://github.com/ARMmbed/mbed-os/commit/6220ca5fd5a0a406cca09f7c6e1ac5fd0c5f022a#diff-332c3b0b50e733b909e94d5fd7ac9c2d5dca17a72d0b32cdafc31c293d9e64c3

closed 2021-02-18 by hugueskamba