ARMmbed/mbed-os #14273: #14132 prohibits any code running on Artemis Thing Plus

idea--list, 2021-02-11:
### Description of defect
No code written for my Artemis Thing Plus board work with Mbed OS V6.7

#### Target(s) affected by this defect ?
Sparkfun Artemis Thing Plus (but all the other Apollo3 based boards might be affected as well)

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.14

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS V6.7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.3.1

#### How is this defect reproduced ? 
Try to compile the official blinky (or any other code) with Mbed OS V6.7
Then roll back #14132 compile again and the code will work.

0xc0170, 2021-02-12:
@hugueskamba Please review

cc @ARMmbed/team-sparkfun

idea--list, 2021-02-12:
Compiled the official blinky and 2-3 other code with this change and code runs again as expected.

idea--list, 2021-02-12:
#14277 fixed the issue

closed 2021-02-12 by idea--list