ARMmbed/mbed-os #14283: Unable to compile anything for NUCLEO_H743ZI2 with mbed-tools

sstelian, 2021-02-13:
### Description of defect

#### Target(s) affected by this defect ?
NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 6.1.0

#### How is this defect reproduced ? 
-create new program: mbed-tools new test
-compile: mbed-tools compile -t GCC_ARM -m NUCLEO_H743ZI2

sstelian, 2021-02-13:
Sorry. I wrote the description between the comment tags. Fixed.

jeromecoutant, 2021-02-15:
Should be corrected by #14275 
@rwalton-arm

0xc0170, 2021-02-15:
@sstelian Thanks for the report. Can you try the latest master now? We fixed the most of the targets now.

sstelian, 2021-02-16:
@0xc0170 Latest master works now. Thank you!

closed 2021-02-16 by sstelian