ARMmbed/mbed-os #14317: CYTFM_064B0S2_4343W: Fails to complete post build operation

hugueskamba, 2021-02-22:
### Description of defect
The post build operation does not complete.

#### Target(s) affected by this defect ?
`CYTFM_064B0S2_4343W`

#### Toolchain(s) (name and version) displaying this defect ?
ARM, GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-6.8.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
```
$ mbed-tools --version
7.1.0
```

#### How is this defect reproduced ? 
Attempt to build `mbed-os-example-blinky` run the following command:
`rm -rf cmake_build/ && mbedtools compile -t gcc_ARM -m CYTFM_064B0S2_4343W`

Outcome:
```
...
Total Static RAM memory (data + bss): 11236(+11236) bytes
Total Flash memory (text + data): 51732(+51732) bytes

[338/338] Generating mbed-os-example-blinky.bin
FAILED: mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/mbed-os-example-blinky.bin 
cd /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/GCC_ARM/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W && /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py sign --build-dir /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/GCC_ARM --m0hex-filename tfm_s.hex --target-name CYTFM_064B0S2_4343W --policy-file-name policy_multi_CM0_CM4_tfm.json --boot-scheme multi_image --cm0-img-id 1 --cm4-img-id 16 --elf /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/GCC_ARM/mbed-os-example-blinky.elf --m4hex /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/GCC_ARM/mbed-os-example-blinky.hex --m0hex /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/COMPONENT_TFM_S_FW/tfm_s.hex
2021-02-22 13:43:16,055 : C : ERROR : Cannot find image with id 1 and type 'BOOT' in the policy file. Check the log for details
2021-02-22 13:43:16,055 : C : ERROR : Image with ID 1 not found in '/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json'. Check the log for details
Traceback (most recent call last):
  File "/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py", line 393, in <module>
    args.func(args)
  File "/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py", line 316, in sign_action
    sign_hex(
  File "/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py", line 229, in sign_hex
    sign_application(notification.debug, tools, m0hex, image_id=cm0_img_id)
  File "/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py", line 254, in sign_application
    + str(image_id) + " is " + hex(address) + ", " + hex(size))
TypeError: 'NoneType' object cannot be interpreted as an integer
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.
```

0xc0170, 2021-02-22:
Which object is `NoneType` ? @ARMmbed/team-cypress can you review the post build hook?

ifyall, 2021-02-23:
@0xc0170, can you assign this item to me? My team owns Mbed OS integrations going forward. Thank you.

I have filed an internal ticket on this for our tracking purposes.

ifyall, 2021-02-23:
@hugueskamba,

Where is the policy_multi_CM0_CM4_tfm.json file that the build is complaining about? It doesn't seem to be in mbed-os but I don't see your reproduction steps documenting creation/management of it.

Thank you,

Ian

RaymondNgun, 2021-02-23:
`cysecuretools -t cy8kit-064b0s2-4343w init ` creates the policy file.  Have a look at.

https://os.mbed.com/platforms/CY8CKIT-064B0S2-4343W/#building-the-mbed-os-blinky-application

hugueskamba, 2021-02-23:
> @hugueskamba,
> 
> Where is the policy_multi_CM0_CM4_tfm.json file that the build is complaining about? It doesn't seem to be in mbed-os but I don't see your reproduction steps documenting creation/management of it.
> 
> Thank you,
> 
> Ian

The policy file is in Mbed OS. Here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json
Attempting to build as specified will show the same error.

hugueskamba, 2021-02-23:
With the old Mbed CLI I get the following error:
```
$ rm -rf BUILD && rm -rf cmake_build/ && mbed compile -t gcc_ARM -m CYTFM_064B0S2_4343W
...
Compile [100.0%]: cy_usbfs_dev_drv_io_dma.c
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
Post-Build: mbed-os-example-blinky
M0 core image file found: ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/COMPONENT_TFM_S_FW/tfm_s.hex.
Policy file found: /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json.
[PSOC6.sign_image] Using policy file: /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json
[ERROR] No module named 'imgtool.main'; 'imgtool' is not a package
[mbed] ERROR: "/usr/local/opt/python@3.9/bin/python3.9" returned error.
       Code: 1
       Path: "/path/to/mbed-os-example-blinky"
       Command: "/usr/local/opt/python@3.9/bin/python3.9 -u /path/to/mbed-os-example-blinky/mbed-os/tools/make.py -t gcc_ARM -m CYTFM_064B0S2_4343W --source . --build ./BUILD/CYTFM_064B0S2_4343W/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

RaymondNgun, 2021-02-23:
@hugueskamba, if you already don't know, this fix is needed for that

https://github.com/LDong-Arm/mbed-os/commit/e5901ecdba1bbb2253d1c0cf1693907a1b668297

hugueskamba, 2021-02-23:
> LDong-Arm@e5901ec

Oh I see. I was not aware of that. Thanks.
However, I am still not sure how to solve the issue in this issue description.

RaymondNgun, 2021-02-23:
@hugueskamba, I'm not sure how that error is encountered.  To recap, that policy file does indeed exist. Are you using cysecuretools 2.1.0?

hugueskamba, 2021-02-23:
> @hugueskamba, I'm not sure how that error is encountered.  To recap, that policy file does indeed exist. Are you using cysecuretools 2.1.0?

The policy file exists and I tried again after I downgraded to cysecuretools version 2.1.0 following your comment on the other issue.

RaymondNgun, 2021-02-24:
@hugueskamba, I don't use mbed-tools but use the older cli. It seems this happens when using mbed-tools.

hugueskamba, 2021-02-24:
I included the fix [here](https://github.com/LDong-Arm/mbed-os/commit/e5901ec) in order to build with the old tools and the post build operation completed successfully.
I proceeded by adding print statements in the [old tools post build script](https://github.com/ARMmbed/mbed-os/blob/master/tools/targets/PSOC6.py) and got the following output:

```
Post-Build: mbed-os-example-blinky
M0 core image file found: ./mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/COMPONENT_TFM_S_FW/tfm_s.hex.
Policy file found: /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json.
[PSOC6.sign_image] Using policy file: /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json

secure_target=cy8ckit-064b0s2-4343w
policy_file=/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json

tools=<cysecuretools.main.CySecureTools object at 0x10813a7f0>
m0hex=./BUILD/CYTFM_064B0S2_4343W/ARM/tfm_s.hex
binf=./BUILD/CYTFM_064B0S2_4343W/ARM/mbed-os-example-blinky.hex
toolchain.target.cm0_img_id=1
toolchain.target.cm4_img_id=16

image_id = 1, binary=./BUILD/CYTFM_064B0S2_4343W/ARM/tfm_s.hex
2021-02-24 19:54:56,837 : C : INFO  : Image for slot BOOT signed successfully! (./BUILD/CYTFM_064B0S2_4343W/ARM/tfm_s.hex)
2021-02-24 19:54:57,113 : C : INFO  : Image for slot UPGRADE signed successfully! (./BUILD/CYTFM_064B0S2_4343W/ARM/tfm_s_upgrade.hex)

image_id = 16, binary=./BUILD/CYTFM_064B0S2_4343W/ARM/mbed-os-example-blinky.hex
2021-02-24 19:54:57,592 : C : INFO  : Image for slot BOOT signed successfully! (./BUILD/CYTFM_064B0S2_4343W/ARM/mbed-os-example-blinky.hex)
2021-02-24 19:54:57,674 : C : INFO  : Image for slot UPGRADE signed successfully! (./BUILD/CYTFM_064B0S2_4343W/ARM/mbed-os-example-blinky_upgrade.hex)

| Module               |     .text |   .data |      .bss |
|----------------------|-----------|---------|-----------|
| [lib]/c_w.l          |  4538(+0) |  16(+0) |   348(+0) |
| [lib]/fz_wm.l        |    26(+0) |   0(+0) |     0(+0) |
| [lib]/libcppabi_w.l  |    44(+0) |   0(+0) |     0(+0) |
| anon$$obj.o          |    84(+0) |   0(+0) |  5120(+0) |
| main.o               |   295(+0) |   0(+0) |     0(+0) |
| mbed-os/cmsis        |  9949(+0) | 168(+0) |  6865(+0) |
| mbed-os/connectivity |   200(+0) |   0(+0) |    52(+0) |
| mbed-os/drivers      |   578(+0) |   0(+0) |     0(+0) |
| mbed-os/hal          |  3093(+0) |   8(+0) |   232(+0) |
| mbed-os/platform     |  8873(+0) |  64(+0) |   628(+0) |
| mbed-os/rtos         |   420(+0) |   0(+0) |     0(+0) |
| mbed-os/targets      | 29327(+0) | 557(+0) |  1738(+0) |
| Subtotals            | 57427(+0) | 813(+0) | 14983(+0) |
Total Static RAM memory (data + bss): 15796(+0) bytes
Total Flash memory (text + data): 58240(+0) bytes

Image: ./BUILD/CYTFM_064B0S2_4343W/ARM/mbed-os-example-blinky.hex
```

I compared it to the print statement in the [new tools post build script](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Cypress/scripts/PSOC6.py) and got the following output:
```
FAILED: mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/CMakeFiles/mbed-post-build-bin-CYTFM_064B0S2_4343W.util 

cd /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W && /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py sign --build-dir /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM --m0hex-filename tfm_s.hex --target-name CYTFM_064B0S2_4343W --policy-file-name policy_multi_CM0_CM4_tfm.json --boot-scheme multi_image --cm0-img-id 1 --cm4-img-id 16 --elf /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/mbed-os-example-blinky.elf --m4hex /path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/mbed-os-example-blinky.hex --m0hex /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/COMPONENT_TFM_S_FW/tfm_s.hex
[PSOC6.sign_image] Using policy file: /path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json

secure_target=cy8ckit-064b0s2-4343w
policy_file=/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json

tools=<cysecuretools.main.CySecureTools object at 0x1080e94f0>
m0hex=/path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/tfm_s.hex
m4hex=/path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/mbed-os-example-blinky.hex
toolchain.target.cm0_img_id=1
toolchain.target.cm4_img_id=16

image_id = 1, binary=/path/to/mbed-os-example-blinky/cmake_build/CYTFM_064B0S2_4343W/develop/ARM/tfm_s.hex

2021-02-24 19:52:56,185 : C : ERROR : Cannot find image with id 1 and type 'BOOT' in the policy file. Check the log for details
2021-02-24 19:52:56,185 : C : ERROR : Image with ID 1 not found in '/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/TARGET_CYTFM_064B0S2_4343W/policy/policy_multi_CM0_CM4_tfm.json'. Check the log for details
Traceback (most recent call last):
[... cut ...]
  File "/path/to/mbed-os-example-blinky/mbed-os/targets/TARGET_Cypress/scripts/PSOC6.py", line 264, in sign_application
    + str(image_id) + " is " + hex(address) + ", " + hex(size))
TypeError: 'NoneType' object cannot be interpreted as an integer
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!
```

Given the input provided to [`sign_application()`](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Cypress/scripts/PSOC6.py#L239) is the similar, I still do not understand why the image cannot be found
with the new tools script.🤔

RaymondNgun, 2021-02-24:
@hugueskamba, it looks like the problem is the argument cm0-img-id is treated as a string. If I update the type of the argument, this problem will be fixed.  E.g. in PSOC6.py, update the following to be type int.

`    sign_subcommand.add_argument(
        "--cm0-img-id", type=int, help="the Cortex-M0 image ID."
    )
    sign_subcommand.add_argument(
        "--cm4-img-id", type=int, help="the Cortex-M4 image ID."
    )`

hugueskamba, 2021-02-24:
> @hugueskamba, it looks like the problem is the argument cm0-img-id is treated as a string. If I update the type of the argument, this problem will be fixed. E.g. in PSOC6.py, update the following to [...]

AH! That does indeed fix the issue! Thanks.

closed 2021-02-25 by 0xc0170