ARMmbed/mbed-os #14320: STM32F4 DFU Mbed 6

tartavull, 2021-02-22:
My application requires flashing the firmware without pressing any buttons on the device. Therefore, I would like to be able to enter DFU mode from the application.

I'm using the code suggested in https://github.com/ARMmbed/mbed-os/issues/8523#issuecomment-500913569

**main.c**
```
#include "mbed.h"
extern "C" void enter_dfu();

int main() {
    DigitalOut usb_dp_pin(USB_OTG_FS_DP, 1) ;
    usb_dp_pin=0;

    enter_dfu();
}
```
**bootloader.S**
```
    AREA asm_func, CODE, READONLY
    EXPORT enter_dfu
enter_dfu
    LDR     R0, =0x40023844 ; RCC_APB2ENR
    LDR     R1, =0x00004000 ; ENABLE SYSCFG CLOCK
    STR     R1, [R0, #0]
     
    LDR     R0, =0x40013800 ; remap ROM at zero
    LDR     R1, =0x00000001 ; SYSCFG_MEMRMP
    STR     R1, [R0, #0]
       
    LDR     R0, =0x1FFF0000 ; ROM BASE
    LDR     SP,[R0, #0]     ; SP @ +0
    LDR     R0,[R0, #4]     ; PC @ +4
    BX      R0              ; Jump to Bootloader
    END
```

I'm using a custom target (which I verified it is working by blinking LEDs)
**custom_targets.json**
```
    "MTS_MDOT_F405RG": {
        "inherits": ["MCU_STM32"],
        "core": "Cortex-M4F",
        "extra_labels_add": ["STM32F4", "STM32F405RG"],
        "is_disk_virtual": true,
        "macros_add": ["HSE_VALUE=26000000"],
        "features_add": ["BOOTLOADER"],
        "device_has_add": ["ANALOGOUT", "MPU"],
        "release_versions": ["2", "5"],
        "device_name": "STM32F405RG",
        "bootloader_suported": true,
	"device_has_remove": ["SERIAL_FC"]
    }
}
```
I'm compiling using ARMC6 with mbed-studio and using 
commit 0c6753bb821612eb72b5f682aed7ffcc98e80a1f (HEAD -> latest, tag: mbed-os-6.7.0-rc1, tag: mbed-os-6.7.0, origin/latest)

The device doesn't enter DFU mode, I would appreciate any pointers.

tartavull, 2021-02-25:
Is there any information I can add here?

0xc0170, 2021-02-25:
@tartavull If you can update the above to use the default template (we parse the information internally). Thanks

@ARMmbed/team-multitech Please review

0xc0170, 2021-02-25:
@ARMmbed/team-st-mcd might help with DFU mode as well?

jeromecoutant, 2021-02-25:
Maybe you can try to add a little delay before setting usb_dp_pin to 0 ?

DheerN, 2021-02-25:
Added a 1s delay before setting usb_dp_pin to 0. Didn't make a difference. Any other solutions?

jeromecoutant, 2021-02-25:
For USB, I am waiting 10ms:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/USBPhy_STM32.cpp#L200

DheerN, 2021-02-25:
10ms delay hasn't worked either.

JojoS62, 2021-02-25:
what about the interrupt vector table, is it necessary to reset the SCB->VTOR ? It is set to ram in platform/source/mbed_sdk_boot.c

tartavull, 2021-04-04:
This can be closed, we got it working. Thanks a lot for the help!

closed 2021-04-04 by tartavull

jeromecoutant, 2021-04-06:
@tartavull maybe you could add few details to help community in the future ? :-)

btply, 2021-05-05:
@tartavull Could you share your solution please? I'm also having the same issue

tartavull, 2021-05-05:
We "solved it" by using some external circuitry.