ARMmbed/mbed-os #14330: mbed-os hard fault in CY8CKIT064B0S2_4343W

saheerb, 2021-02-23:
### Description of defect

mbed-os-example-blinky (and greentea tests) are hardfaulting for CY8CKIT064B0S2_4343W target.

#### Target(s) affected by this defect ?

https://os.mbed.com/platforms/CY8CKIT-064B0S2-4343W/

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM (tested only with this)

#### What version of Mbed-os are you using (tag or sha) ?
The issue is happening with:
[current master](https://github.com/ARMmbed/mbed-os/commit/8340ea2d2b39d12a5220928b30c4b3dc90fd5dff)
[mbed-os-6.7.0 ](https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-6.7.0)

The issue is not reproducible with [mbed-os.6.6.0 ](https://github.com/ARMmbed/mbed-os/releases/tag/mbed-os-6.6.0)

```
++ MbedOS Fault Handler ++
FaultType: HardFault
Context:
R   0: 011A0FF9
R   1: 00000000
R   2: 00000E63
R   3: 00000008
R   4: 10022D84
R   5: 10022D84
R   6: AFFFFFDF
R   7: 00000000
R   8: FFFFDFFF
R   9: FFE8FD7F
R  10: 080D0000
R  11: 00000000
R  12: 00000000
SP   : 080DFFD8
LR   : 1001D271
PC   : 1001D2DE
xPSR : 21000000
PSP  : F9B79FFC
MSP  : 080DFFB8
CPUID: 410FC241
HFSR : 80000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000000
DFSR : 00000002
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: MSP
-- MbedOS Fault Handler --
++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x1001D2DE
Error Value: 0x8004CD4
Current Thread: <unnamed> Id: 0x0 Entry: 0x2A00400A StackSize: 0x18894923 StackMem: 0x4A27E003 SP: 0x80DFFD8
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=CY8CKIT064B0S2_4343W
-- MbedOS Error Info --
```

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
cysecuretools-2.0.0

_Note: With cysecuretools-2.1.0, I encounter this issue: https://github.com/ARMmbed/mbed-os/issues/14153_

#### How is this defect reproduced ? 
(install cysecuretools-2.0.0)

- git clone https://github.com/ARMmbed/mbed-os-example-blinky
- cd mbed-os-example-blinky
- mbed deploy
- cd targets/TARGET_Cypress/TARGET_PSOC6/CY8CKIT064B0S2_4343W
- cysecuretools -t cy8ckit-064b0s2-4343w init
- cysecuretools -t cyb06xxa -p policy/policy_single_CM0_CM4.json re-provision-device
- cysecuretools -t cy8ckit-064b0s2-4343w create-keys
- mbed compile -m CY8CKIT064B0S2_4343W -t GCC_ARM -v
- Flash the binary built

Could also reproduce in greentea:

- git clone https://github.com/ARMmbed/mbed-os
- cd targets/TARGET_Cypress/TARGET_PSOC6/CY8CKIT064B0S2_4343W
- cysecuretools -t cy8ckit-064b0s2-4343w init
- cysecuretools -t cyb06xxa -p policy/policy_single_CM0_CM4.json re-provision-device
- cysecuretools -t cy8ckit-064b0s2-4343w create-keys
- mbed test -m CY8CKIT064B0S2_4343W -t GCC_ARM -n "*printf*" -v

0xc0170, 2021-02-23:
@ARMmbed/team-cypress Please review

saheerb, 2021-03-05:
@ifyall just wanted to check if you were able to look at this?

0xc0170, 2021-03-05:
> @ifyall just wanted to check if you were able to look at this?

We got code freeze in a week, it would be nice to resolve this.

saheerb, 2021-04-06:
@ifyall gentle reminder to check if this has been looked into.

ifyall, 2021-04-06:
@saheerb, we are working on generating a PR that should address this. Goal is end of this week or mid-next.

saheerb, 2021-05-04:
@ifyall any update on this?