ARMmbed/mbed-os #14335: Unable to debug 6.6 onwards

chrissnow, 2021-02-23:
### Description of defect
Debugging the XDOT_L151CC does not seem possible anymore, using a J-Link and SWD program succeeds but it disconnects immediately on connect. 6.5 works fine.

#### Target(s) affected by this defect ?
XDOT_L151CC

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
uVision or CLI build

#### How is this defect reproduced ? 
attempt to debug blinky 6.6 onwards with a j-link

Could affect other targets and debuggers.

0xc0170, 2021-02-23:
@chrissnow Have you identified any changes to STM32L1 family or this target in particular?

cc @ARMmbed/team-st-mcd

chrissnow, 2021-02-23:
@0xc0170 I did look and couldn't see anything obviously of interest in STM32L1 between releases.
I suspect it may not be L1 specific, If i have time tomorrow I will try a different target.

jeromecoutant, 2021-02-24:
Hi
Issue is known and corrected since #14032

jeromecoutant, 2021-02-24:
My mistake...

> Issue is known and corrected since #14032

Except for STM32L1...
I need to update something else...
You can remove these lines:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3312-L3314

jeromecoutant, 2021-02-24:
Hi again
Could you check #14337 ?

0xc0170, 2021-03-02:
@chrissnow any update?

chrissnow, 2021-03-02:
@0xc0170  Apologies for the delay, I was away on site late last week.

I haven't tested the change above, but the changes in #14337 wont build for me (comments left).

jeromecoutant, 2021-03-09:
Let's retry with the new master branch ?
Thx

chrissnow, 2021-03-12:
@jeromecoutant all good now in master :-)

closed 2021-03-12 by chrissnow