ARMmbed/mbed-os #14353: Setting BufferedSerial pin to NC throws runtime assertion

mark-psl, 2021-02-25:
### Description of defect

A runtime assertion happens on FRDM-K64F when assigning a pin name as "NC".  Specifically this happens on BufferedSerial() when assigning the Rx pin to be NC.  This is in conflict with the mbed-os API document on BufferedSerial().

`TX and RX pins - you can specify either pin as Not Connected (NC) for simplex (unidirectional) communication or both as valid pins for full duplex (bidirectional) communication.`

The following is the assertion:

`++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: pin != (PinName)NC
Location: 0x85F7
File: ./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/api/pinmap.c+27
Error Value: 0x0
Current Thread: main Id: 0x1FFF21DC Entry: 0x91CD StackSize: 0x1000 StackMem: 0x1FFF0C30 SP: 0x1FFF1AD0 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=K64F
-- MbedOS Error Info --`

It appears the assertion happens in the pin_function() routine called by the initialization code in serial_api.c for this target.  
In other targets where this is not an issue (eg. FRDM-KL46Z) the call to pin_function() is first checked to ensure the pin is not NC.  
In the target in question (FRDM-K64F) the pin_function() calls are done without the check even through if(!=NC) checks are done just below it to enable the Tx and Rx pins.  

Starting at line 66 it currently looks like:

```
pin_function(pinmap->tx_pin, pinmap->tx_function);
pin_function(pinmap->rx_pin, pinmap->rx_function);

if (pinmap->tx_pin != NC) {
    UART_EnableTx(uart_addrs[obj->serial.index], true);
    pin_mode(pinmap->tx_pin, PullUp);
}
if (pinmap->rx_pin != NC) {
    UART_EnableRx(uart_addrs[obj->serial.index], true);
    pin_mode(pinmap->rx_pin, PullUp);
}
```
I moved the pin_function() calls into the if(!=NC) sections for Tx and Rx as such:

```
if (pinmap->tx_pin != NC) {
    pin_function(pinmap->tx_pin, pinmap->tx_function);
    UART_EnableTx(uart_addrs[obj->serial.index], true);
    pin_mode(pinmap->tx_pin, PullUp);
}
if (pinmap->rx_pin != NC) {
    pin_function(pinmap->rx_pin, pinmap->rx_function);
    UART_EnableRx(uart_addrs[obj->serial.index], true);
    pin_mode(pinmap->rx_pin, PullUp);
}

```
This corrects the issue and complies with the documentation on BufferedSerial() in the mbed-os API documentation.

#### Target(s) affected by this defect ? 
FRDM-K64F

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio v1.3.1

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os v6.8

It also appears to be in v6.7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio v1.3.1

#### How is this defect reproduced ? 
Set the Rx pin for BufferedSerial() to NC.  This will compile and link cleanly but throw a runtime assertion and restart the board.

0xc0170, 2021-03-02:
Thanks for the report. @mark-psl can you please send a pull request fixing this?

mark-psl, 2021-03-02:
Sure.  Not sure how to do that correctly for this issue.  Do you have a branch that I can refer to?

0xc0170, 2021-03-05:
Create your own branch with a fix and send it via "new pull request"

mark-psl, 2021-03-05:
I'm unable to create a branch from master.  I can only filter on this repo.

ladislas, 2021-03-05:
you need to fork the repo first.

mark-psl, 2021-03-05:
Done.  PR#14387 submitted.  I'm a noob at this so thanks for your patience.