ARMmbed/mbed-os #14360: Tcp server mbed

nadaanas1988, 2021-03-01:
Hello everyone.

I have a radio system which is accepting GPS data from an external TCP/IP server. My radio system is connecting via a VLAN network.

I'm using ublox- C027 as a TCP server to send the GPS data to my radio but I have an issue with the connection. The VLAN is already enabled on my mbed but when my server try to accept the connection from the radio(TCP client) the accept function returning the value -3001.

Here is my mbed program:

`#include "mbed.h"
#include "gnss.h"
#include<string>
#include "EthernetInterface.h"
#include "rtos.h"
#include "TCPServer.h"
#include "TCPSocket.h"

// LEDs
DigitalOut led_1(P2_4);
DigitalOut led_2(P2_3);
DigitalOut led_3(P1_21);
DigitalOut led_4(P1_24);

// Static IP network variables
static const char*          mbedIP       = "10.5.248.45";  //IP 
static const char*          mbedMask     = "255.255.248.0";  // Mask
static const char*          mbedGateway  = "10.5.248.1";    //Gateway

static const char* recvIP = "10.5.248.121";

int main()
{
    pc.baud(115200);
    pc.printf("TCP server example\r\n");
    
    EthernetInterface eth;
    eth.set_network(mbedIP , mbedMask, mbedGateway );// my device address
    //eth.connect();
    
    if (0 != eth.connect())
        {
            
            pc.printf("Ethernet not connected\n\r");
            led_1=!led_1;
            return -1;
        }
    else
        {
            pc.printf("Connected Ethernet \n\r");
            led_2=!led_2; 
            
        }
            
    
    //pc.printf("The Server IP address is '%s'\r\n", eth.get_ip_address());
        
    TCPServer srv;
    TCPSocket client_sock[MAXCLIENTS];
    SocketAddress client_addr[MAXCLIENTS];
    
    int connectionStatus[MAXCLIENTS];
    
    char *buffer = new char[256];
    
    // Open the server on ethernet stack 
    srv.open(&eth);
    
    // Bind the HTTP port (TCP 80) to the server 
    srv.bind(10110);
    
    // Can handle x simultaneous connections 
    srv.listen(MAXCLIENTS);
    srv.set_blocking(false);
    //client_sock[0].set_blocking(false); 
 
    int numClient = 0;
    int rcount = 0;
    int scount = 0;
    
    // initialize connection Status for client
    for(int i=0;i<MAXCLIENTS;i++)
        connectionStatus[i] = 0;
    
    while(1)
    {
        for(int i=0;i < MAXCLIENTS;i++)
        {
            //pc.printf("coonection : %i/n/r",connectionStatus[i]);
            // if not connected accept connection
            if (connectionStatus[i] == 0)
            {
                pc.printf("accept : %i\n\r",srv.accept(&client_sock[i], &client_addr[i]));
                if(srv.accept(&client_sock[i], &client_addr[i]) == 0)
                {  
                    connectionStatus[i] = 1;
                    client_sock[i].set_blocking(false); 
                    //srv.set_blocking(false);         
                    numClient++;                    
                    pc.printf("Client Accepted %s:%d,%d\r\n", client_addr[i].get_ip_address(),client_addr[i].get_port(),numClient);
                    led_3=!led_3; 
                }
            }
            
            // if connected        
            if (connectionStatus[i] == 1)
            {                                             
                rcount = client_sock[i].recv(buffer, sizeof(buffer));
                if(rcount > 0)
                {
                    sprintf(buffer, "Hello Client %s:%d\r\n", client_addr[i].get_ip_address(),client_addr[i].get_port());
                    scount = client_sock[i].send(buffer, strlen(buffer));
                    led_4=!led_4;
                }
                if(rcount == 0) // client disconnected
                {
                    connectionStatus[i] = 0;
                    pc.printf("Client Disconnected %s:%d\r\n", client_addr[i].get_ip_address(),client_addr[i].get_port());
                    client_sock[i].close();
                    numClient--;                
                }                                                        
                if(scount < 0) // unable to send data
                {
                    connectionStatus[i] = 0;
                    pc.printf("Connection lost %s:%d\r\n", client_addr[i].get_ip_address(),client_addr[i].get_port());
                    client_sock[i].close();
                    numClient--;                
                }                                                        
            }
        }
 
        wait(0.5f);
    }
 
}`

I'm using mbed os 5.15.6

Please, help me as I tried so many TCP server programs but nothing working with this radio system. 

Your help will be really apprecieted.

Regards
Nada

nadaanas1988, 2021-03-02:
Thank you and sorry for that. .

I didn' know about this as it's my first time to raise an issue on github.

Sorry again

Regards
Nada

0xc0170, 2021-03-02:
As this is question rather than an actual technical issue, please report this to our forum https://forums.mbed.com/

I'll close this.

closed 2021-03-02 by 0xc0170

nadaanas1988, 2021-03-02:
Dear Martin,

I tried to post many times on mbed forum about the VLAN and using the VLAN
also about this kind of issue but no reply from the mbed team.

This is really important to move on with my project which I have a deadline
for it.

Regards
Nada