ARMmbed/mbed-os #14365: Ethernet UDP connect() gives wrong nsapi_size_or_error_t fo recv()

sstaub, 2021-03-02:
### Description of defect

See also [https://forums.mbed.com/t/ethernet-receive-problem/12217/19](https://forums.mbed.com/t/ethernet-receive-problem/12217/19)

On UDP socket I'm using the `connect()` function for sending data with `send()`
This does not work in conjunction with the `recv()` function, I get always -3001 as a result of `nsapi_size_or_error_t`
But the packets are arrived and I can make them visible with trick described in the forum post.
When I comment out the `connect()` function and use `sendTo()` instead of `send()` the `nsapi_size_or_error_t` is correct,
see the code at the end of this post.

```cpp
#include "mbed.h"
#include "EthernetInterface.h"

#define UDP_RECEIVE_BUFFER 255

// local network settings
const uint8_t LOCAL_IP[4] = {10, 101, 1, 201};
const uint8_t SUBNET[4] = {255, 255, 0, 0};

// grandMA3 network settings
const uint8_t eosIP[4] = {10, 101, 1, 100};
const uint16_t eosUdpRxPort = 8000; // receive port of the console
const uint16_t eosUdpTxPort = 8001; // send port of the console

EthernetInterface eth;
UDPSocket udp;

SocketAddress localIP(LOCAL_IP, NSAPI_IPv4);
SocketAddress subnet(SUBNET, NSAPI_IPv4);
SocketAddress send(eosIP, NSAPI_IPv4, eosUdpRxPort);
SocketAddress receive;

int main() {
	eth.set_network(localIP, subnet, NULL);
	eth.set_blocking(false);
	eth.connect();

	udp.open(&eth);
	udp.set_blocking(false);
	udp.connect(send);
	udp.bind(eosUdpTxPort);

	while (true) {
		static char buffer[UDP_RECEIVE_BUFFER];
		nsapi_size_or_error_t size = udp.recv(buffer, sizeof(buffer));
		if (size > 0) {
			printf("NSAPI code: %d\n", size);
			printf("message: %s\n\n", buffer);
			udp.send(buffer, size); // echo
			memset(buffer, 0, UDP_RECEIVE_BUFFER);
			}
		}
	}
```
I would expect that the use of `connect()` doesn't block the correct `nsapi_size_or_error_t` output.

#### Target(s) affected by this defect ?
STM32-NucleoF767ZI

#### Toolchain(s) (name and version) displaying this defect ?
ARM 6.1.4

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.8.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.3.1
macOS Big Sur 11.2.2

#### How is this defect reproduced ? 

Sending an UDP packet witch PacketSender app gives no output.
This code works:

```cpp
#include "mbed.h"
#include "EthernetInterface.h"

#define UDP_RECEIVE_BUFFER 255

// local network settings
const uint8_t LOCAL_IP[4] = {10, 101, 1, 201};
const uint8_t SUBNET[4] = {255, 255, 0, 0};

// grandMA3 network settings
const uint8_t eosIP[4] = {10, 101, 1, 100};
const uint16_t eosUdpRxPort = 8000; // receive port of the console
const uint16_t eosUdpTxPort = 8001; // send port of the console

EthernetInterface eth;
UDPSocket udp;

SocketAddress localIP(LOCAL_IP, NSAPI_IPv4);
SocketAddress subnet(SUBNET, NSAPI_IPv4);
SocketAddress send(eosIP, NSAPI_IPv4, eosUdpRxPort);
SocketAddress receive;

int main() {
	eth.set_network(localIP, subnet, NULL);
	eth.set_blocking(false);
	eth.connect();

	udp.open(&eth);
	udp.set_blocking(false);
	udp.bind(eosUdpTxPort);

	while (true) {
		static char buffer[UDP_RECEIVE_BUFFER];
		nsapi_size_or_error_t size = udp.recv(buffer, sizeof(buffer));
		//nsapi_size_or_error_t size = udp.recvfrom(&receive, buffer, sizeof(buffer));
		if (size > 0) {
			printf("NSAPI code: %d\n", size);
			printf("message: %s\n\n", buffer);
			udp.sendto(send, buffer, size); // echo
			memset(buffer, 0, UDP_RECEIVE_BUFFER);
			}
		}
	}
```

0xc0170, 2021-03-05:
cc @ARMmbed/mbed-os-connectivity

paul-szczepanek-arm, 2021-03-17:
Thank you for reporting it. We've scheduled this work for this sprint so I'll get to it shortly.

paul-szczepanek-arm, 2021-03-19:
I cannot reproduce your problem. I can send udp packets fine with blocking and not blocking version.

Here's a few things to keep in mind though. If you use a non blocking mode you need to wait until the  `eth.connect();` actually connects, you can use get_connection_status() to see if it's ready.

Secondly, please show me the wireshark capture. I suspect your packet that  you're sending is filtered. Open InternetDatagramSocket.cpp and go to InternetDatagramSocket::recvfrom. Before the continue here:
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/netsocket/source/InternetDatagramSocket.cpp#L111
put
```
            printf("filtered remote %s received %s\r\n", _remote_peer.get_ip_address(), address->get_ip_address());
            printf("filtered remote port %d received port %d\r\n", _remote_peer.get_port(), address->get_port());
```
And see if your packet isn't being received after all.

sstaub, 2021-03-19:
There is no filtering, blocking or anything else, The example in the beginning of this issue doesn't work but it should. The example at the end works fine.
In the forum post I made a modification to the first example:
```cpp
while (true) {
		static char buffer[UDP_RECEIVE_BUFFER];
		nsapi_size_or_error_t size = udp.recv(buffer, sizeof(buffer));
		if (buffer[0] != 0) {
			printf("message size: %d\n", size);
			printf("OSC message: %s\n", buffer);
			memset(buffer, 0, UDP_RECEIVE_BUFFER);
			}
		}
```
with following output:
[image]
You can see that I get always -3001 for `nsapi_size_or_error_t`but the buffer is filled with the correct message.

paul-szczepanek-arm, 2021-03-19:
You are showing me the contents of the buffer, I don't know who wrote that buffer, whether it's the stack or just contents from previous run or what. And the stack can write to the buffer even if it errors. Please put the prints in where I suggested and see if it's not being filtered after all.

sstaub, 2021-03-19:
The content of the buffer comes from packet sender app. The problem is that the buffer is filled but I get an error -3001 instead of the size of the received data. I don't understand what you mean with the stack can write to the buffer, but in this case there should no error. The buffer is always cleared with memset() after data is printed.

paul-szczepanek-arm, 2021-03-19:
I'm trying my best to help you. Please humour me and put those two lines in the code, it won't take you long and will help use debug your problem.

sstaub, 2021-03-19:
```cpp
#include "mbed.h"
#include "EthernetInterface.h"

#define UDP_RECEIVE_BUFFER 255

// local network settings
const uint8_t LOCAL_IP[4] = {10, 101, 1, 201};
const uint8_t SUBNET[4] = {255, 255, 0, 0};

// grandMA3 network settings
const uint8_t eosIP[4] = {10, 101, 1, 100};
const uint16_t eosUdpRxPort = 8000; // receive port of the console
const uint16_t eosUdpTxPort = 8001; // send port of the console

EthernetInterface eth;
UDPSocket udp;

SocketAddress localIP(LOCAL_IP, NSAPI_IPv4);
SocketAddress subnet(SUBNET, NSAPI_IPv4);
SocketAddress send(eosIP, NSAPI_IPv4, eosUdpRxPort);
SocketAddress receive;

int main() {
	eth.set_network(localIP, subnet, NULL);
	eth.set_blocking(false);
	eth.connect();

	udp.open(&eth);
	udp.set_blocking(false);
	udp.connect(send);
	udp.bind(eosUdpTxPort);

	while (true) {
		static char buffer[UDP_RECEIVE_BUFFER];
		nsapi_size_or_error_t size = udp.recv(buffer, sizeof(buffer));
		if (buffer[0] != 0) {
			printf("message size: %d\n", size);
			printf("OSC message: %s\n", buffer);
			memset(buffer, 0, UDP_RECEIVE_BUFFER);
			}
		}
	}
```

paul-szczepanek-arm, 2021-03-19:
The reason I'm suspecting it's being filtered is that it fits what you say about the data being in the buffer, the stack will receive the data, write it to the buffer then decide it's not what it expected, try to receive again and since there is nothing more pending it will return the "would block" error.

sstaub, 2021-03-19:
I think you are on the right way. That behavior is caused off using `udp.connect(send);` in the first example.

paul-szczepanek-arm, 2021-03-19:
If you add those two lines you will know for sure.

pan-, 2021-03-24:
@sstaub I'm closing this. Please reopen if the issue wasn't caused by filtering.

sstaub, 2021-03-24:
I don't know why this issue is closed, it persists and it seems that Mbed team is not able or willing to solve the problems with Ethernetstack.

paul-szczepanek-arm, 2021-03-24:
Apologies. I asked you whether or not your packet is being filtered - you didn't reply so I assumed that was the indeed the case.

What was the effect of inserting the prints? What port is expected vs what port is used?

sstaub, 2021-03-24:
I don't understand what you mean with 'filtered', by a switch? The Nucleo board is directly connected to my computer.
The prints are for debugging the message. The sending port is 8000, the receiving port is bind to 8001.

paul-szczepanek-arm, 2021-03-24:
By filtered I mean by this code right here:
https://github.com/ARMmbed/mbed-os/blob/master/connectivity/netsocket/source/InternetDatagramSocket.cpp#L111
when the port is different from the one you "connected" to the packet will be filtered by the line I'm pointing to (click the link above). If you put the print lines I asked you to put there you will see what port originated the packet vs what port you are "connected" to.

sstaub, 2021-03-24:
Which print lines?

paul-szczepanek-arm, 2021-03-24:
> Open InternetDatagramSocket.cpp and go to InternetDatagramSocket::recvfrom. Before the continue here: [...]

The message above.

sstaub, 2021-03-25:
Inserted the line and using following code:
```cpp
#include "mbed.h"
#include "EthernetInterface.h"

#define UDP_RECEIVE_BUFFER 255

// local network settings
const uint8_t LOCAL_IP[4] = {10, 101, 1, 201};
const uint8_t SUBNET[4] = {255, 255, 0, 0};
const uint16_t LOCAL_PORT = 8001;

// remote network settings
const uint8_t REMOTE_IP[4] = {10, 101, 1, 100};
const uint16_t REMOTE_PORT = 8000;

EthernetInterface eth;
UDPSocket udp;

SocketAddress localIP(LOCAL_IP, NSAPI_IPv4);
SocketAddress subnet(SUBNET, NSAPI_IPv4);
SocketAddress send(REMOTE_IP, NSAPI_IPv4, REMOTE_PORT);

int main() {
	eth.set_network(localIP, subnet, NULL);
	eth.set_blocking(false);
	eth.connect();

	udp.open(&eth);
	udp.set_blocking(false);
	udp.connect(send);
	udp.bind(LOCAL_PORT);

	while (true) {
		static char buffer[UDP_RECEIVE_BUFFER] = {0};
		time_t seconds = time(NULL);
		nsapi_size_or_error_t size = udp.recv(buffer, sizeof(buffer));
		if ((size > 0) || (buffer[0] != 0)) {
			printf("Packet received at %s", ctime(&seconds));
			printf("NSAPI code: %d\n", size);
			printf("message: %s\n\n", buffer);
			udp.send(buffer, strlen(buffer)); // echo
			memset(buffer, 0, UDP_RECEIVE_BUFFER);
			}
		}
	}
```
gives following output:
[image]
I don't understand why filtered.
when commenting out in main the lines
```cpp
udp.connect(send);
udp.send(buffer, strlen(buffer)); // echo
```
i get following result:
[image]

that is packet sende app I use for test:
[image]

paul-szczepanek-arm, 2021-03-25:
It's filtered because it originates from port 53737 as is shown on your screenshot and it's connected to port 8000.
Either send from port 8000 or open another socket and connect it to port 53737.

pan-, 2021-03-25:
From linux manual about [connect](https://man7.org/linux/man-pages/man2/connect.2.html): 

```
If the socket sockfd is of type SOCK_DGRAM, then addr is the
address to which datagrams are sent by default, and the only
address from which datagrams are received.
```

The address parameter contains the IP address **and** the port. If you connect your UDP socket to `SocketAddress send(REMOTE_IP, NSAPI_IPv4, REMOTE_PORT)` then recv filters packet originating from a different address.  
In your case the IP address is correct but the port is not (`53737` instead of `8000`).

sstaub, 2021-03-25:
Ok, meanwhile I found an explanation also on Keil documentation. Sorry, it was a missunderstanding BSD sockets.

closed 2021-03-25 by sstaub