ARMmbed/mbed-os #14370: UDPSocket::sendto() returns a positive integer even if the packet is never sent

Albis-dev, 2021-03-03:
### Description of defect

- Main Issue

**[UDPSocket::sendto()](https://os.mbed.com/docs/mbed-os/v6.7/mbed-os-api-doxy/class_internet_datagram_socket.html#a4a7ab60f016a5d62aef8175464a77198) returns a positive integer even if the packet is never sent. I believe this is related to ARP request.**

- Description

For example, consider the following function.

```
void databomb {
    for (int i=0; i<100; i++) {
        report(to_string(i)); // wraps UDPSocket::sendto()
    }
}
```

Calling `databomb()` right after the necessary initialization prints number 0 to 99 to my console.
However letting the target do nothing for 5 exact minutes and then calling `databomb()` yields a random number between 70 and 80 to 99. 
Debugger shows that missing numbers are actually going through UDPSocket::sendto() and it returns bytes sent as it should be.

After examining the call chain I believe the lwIP stack behaves something like this way.
```
if querying ARP table failed
    if ARP request is already sent
        return immidiately with ERR_OK
    else 
        send ARP request and return immidiately with ERR_OK
else
    send the UDP packet to indexed address and return immidiately with ERR_OK
```

Sure enough, `ARP_MAXAGE` is `300` by default and changing this value affects the time it takes to reproduce the behavior. 
See [lwIP Options](https://os.mbed.com/docs/mbed-os/v6.7/mbed-os-api-doxy/group__lwip__opts__arp.html)

#### Target(s) affected by this defect ?
My custom target(named `EL3K` in `custom_targets.json`) extending `NUCLEO_F429ZI`
(STM32F429ZET + [DP83848CVV](https://www.ti.com/store/ti/en/p/product/?p=DP83848CVV/NOPB))

#### Toolchain(s) (name and version) displaying this defect ?
`arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`

#### What version of Mbed-os are you using (tag or sha) ?
`mbed-os-6.7.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
`mbed-cli 1.10.5`

#### How is this defect reproduced ? 
[Link to full user code/config I used](https://github.com/Albis-dev/udpsend)
1. Build and flash the code. Wait until the target fully initializes.
2. Wait for 5 minutes.
3. Send ascii string "databomb" to a target via UDP socket

I was able to reproduce it with my custom board. However, with `NUCLEO_F767ZI` it works without an issue.
I haven't tested with `NUCLEO_F429ZI` since I do not have it at the moment.

Any help would be appreciated. Thanks!

Albis-dev, 2021-03-04:
Did some investigation and I'd like to correct few things in the previous post.

### lwIP stack behavior

> After examining the call chain I believe the lwIP stack behaves something like this way.
>
> ```
> if querying ARP table failed
>     if ARP request is already sent
>         return immidiately with [...]

Actually It's more like this.

```
if there's no matching entry in ARP table
    if ARP request is already sent
        queue current packet buffer and return immidiately with ERR_OK
        // to send out entire queue once the address is resolved
    else 
        send ARP request and return immidiately with ERR_OK
else 
    send the UDP packet to indexed address and return immidiately with ERR_OK
```
See [etharp_output()](https://github.com/ARMmbed/mbed-os/blob/76b2902f48197f95a0c7f5e19b748cc9dbfb052e/connectivity/lwipstack/lwip/src/core/ipv4/lwip_etharp.c#L875-L891) and [etharp_query()](https://github.com/ARMmbed/mbed-os/blob/76b2902f48197f95a0c7f5e19b748cc9dbfb052e/connectivity/lwipstack/lwip/src/core/ipv4/lwip_etharp.c#L992-L1082).

Also note that [`ARP_QUEUEING` is set to 0 by default](https://github.com/ARMmbed/mbed-os/blob/76b2902f48197f95a0c7f5e19b748cc9dbfb052e/connectivity/lwipstack/lwip/src/include/lwip/opt.h#L655). 

[So packets may be dropped](https://github.com/ARMmbed/mbed-os/blob/76b2902f48197f95a0c7f5e19b748cc9dbfb052e/connectivity/lwipstack/lwip/src/core/ipv4/lwip_etharp.c#L1067).

"Bursty" operations (like `databomb()`) will not work as intended once the ARP cache expires.

### Defect reproduction
> I was able to reproduce it with my custom board. However, with `NUCLEO_F767ZI` it works without an issue.

I was wrong and can confirm that the issue also persists on `NUCLEO_F767ZI`.  To reproduce (in a less tedious way), try these steps. I recommend [PacketSender](https://packetsender.com/) for this.

[Link to full user code/config I used](https://github.com/Albis-dev/udpsend)

    1. Edit `ARP_MAXAGE` to 10
    2. Build and flash the code.
    3. Wait for the "UDPSocket Initilized" message from the target.
    4. Wait for 10 seconds.
    5. Send ascii string "databomb" to a target (192.168.1.135:8001 by default) via UDP.

### Action needs to be taken
There are several workarounds such as making ARP table to never expire or setting `ARP_QUEUEING' and `ARP_QUEUE_LEN` accordingly. But the fundamental problem is still there. I think `sendto()` needs some kind of safety mechanism to prevent silent packet drops.

I have very little programming experience so preparing a fix all by myself is a bit daunting task to me. Like, modifying lwIP codes are allowed? Maybe @kjbracey-arm can give some directions?

kjbracey-arm, 2021-03-04:
UDP is not inherently reliable - silent packet drops are always possible. Protocols have to be able to cope with that. All other error handling is designed starting from that position - we don't _have to_ report errors.

Having said that, some local problems should be detectable. I think ARP failure can show up with some EHOSTUNREACH type errors in most full IP stacks.

But as packets are queued for ARP resolution, the error wouldn't be on the actual packets that failed delivery - you'd get an error on 1 or more subsequent send requests. You'd get a pattern like this:

* packet queued, OK response, resolution attempted, never resolved, ultimately dropped
* packet queued, OK response, waiting for previous resolution to complete, ultimately dropped
* packet not queued, EHOSTUNREACH response (as we've noted the ARP failures and briefly block new resolution attempts)
* packet not queued, EHOSTUNREACH response
* packet queued, OK response, resolution attempted, never resolved, ultimately dropped
* ...

So the error can be generated, and it does mean this packet failed, but lack of error is no assurance that a packet worked.

Now, maybe lwIP doesn't do this, but if not, I wouldn't say it's a bug, it's a missing feature.

The error handling can be refined further by also having a way to detect _some_ non-local drops, by having ICMP errors set an error indication on a connected UDP socket. Again, some stacks can do this.

But, again, as these non-local errors would never be reliable, protocol can't rely on them. And on top of that, even if you DID get them, most such errors should normally be treated as "soft" according to guidelines. You should retry anyway. In which case apps should just be ignoring error returns on `send` and just retry after a backoff timeout if they don't get a reply. Only a few errors like "port unreachable" or "no such socket" should be viewed as hard and trigger giving up.

Looking at the standard "ping" command for example, it will print out any "host unreachable" errors it gets, but they don't affect the basic packet flow or retry patterns - it keeps going regardless.

Anyway, if you want the error-for-ARP-failure feature, I suggest raising it with lwIP developers. I haven't double-checked the code to see if there is something in there - I'm assuming your pseudocode is accurate.

Albis-dev, 2021-03-04:
Thanks for your time and effort in writing this detailed explanation @kjbracey-arm ! It really enlightened me about UDP and the IP stack implementation. 

So if I understood correctly, what you're saying is...

1. Silent packet drop can always happen. UDP is unreliable protocol. If I need some kind of reliability, I should avoid UDP.
2. Thus, the current UDPSocket `send` call has no problem. 
3. ARP is also unreliable. Not providing extra information about ARP resolution is completely valid choice. It's just a missing feature.

Now I remember my college professor saying "UDP doesn't guarantee anything". 😅 

I'm closing this issue.

closed 2021-03-04 by Albis-dev

JojoS62, 2021-03-04:
there is a setting for ARP Queueing in lwip/opt.h, the default is '30'. Even if set to 0, it will queue at least one ARP.
But for STM targets, the lwip-mem setting is very tight, only 2310 bytes, so there maybe not enough memory in this pool. You can enlarge this memory pool by setting ```"lwip.mem-size": 40960``` to the amount of memory that you can spend.

kjbracey-arm, 2021-03-04:
If you used TCP instead, then that is defined as "reliable", which totally changes the equation, and hence API error handling design. There "ok" on a send indicates "I've definitely queued this, and will do my utmost to deliver it", and the API tries its best to indicate subsequent problems.

If it fails to deliver something accepted by a send, it will retry indefinitely until giving up on the connection altogether - either the data gets acknowledged, or it's temporarily blocked by flow control (we're getting responses saying "please wait"), or the connection is broken. (And you'll get an error indication on a subsequent send to say it is broken, which tells you some data may have been lost).

Any queued data should be drained in a blocking socket close (or shutdown) and any failure to deliver already-queued data should be shown in the return code of the close.

So no errors on any send or the close is your guarantee that the data reached the destination.

Although that's _still_ not quite the whole story, because it only means the data reached the TCP level. It could be you finished handshake with a firewall, but it didn't reach the end machine, or it could be you reached the kernel of the destination machine, but the app crashed before reading all the data from its socket. Hence TCP protocols still have explicit "ok" responses to show the app actually processed the request.

TCP lets you ditch the need for app-level retries, but you still normally want an app-level ack. (Unless you're just streaming audio/video or something).