ARMmbed/mbed-os #14371: Adding QSPI feature to mbed-os

kvkc97, 2021-03-03:
Hi,

I am using mbed-os to program STM32WB55RG using MBED CLI2 tool. I have imported the lorawan example code available in the site. The link is added below.
https://github.com/ARMmbed/mbed-os-example-lorawan
I added a custom target in MBED CLI2 and compiled the code successfully. Now I want to add QSPI feature to it. I added QSPI in the "device_has" field in the custom_targets.json file. While compiling the following error is generated. 
[image]
Please suggest any solution.

Regards
Keerthi

0xc0170, 2021-03-05:
What target is it ? `STM32WB55RG ` is not the target name, is this NUCLEO_WB55RG ?

It looks like that target does not have QSPI supported , please verify

kvkc97, 2021-03-05:
This is a custom board which is being designed using STM32WB55RG. I want to add this board to MBED CLI2. The uC has QSPI support and so I want to enable it in the custom board also.

0xc0170, 2021-03-08:
Use `device_has.add` to add QSPI in your custom target json file. And prode QSPI implementation where `qspi_t` is defined. That should fix the error.

jeromecoutant, 2021-03-08:
> I am using mbed-os to program STM32WB55RG using MBED CLI2 tool. I have imported the lorawan example code available in the site. The link is added below.

Note that STM32WB55 is BLE, not LORA...
And I agree QSPI is supported by STM32WB family, but not implemented yet in mbed.

If you want to use LORAWAN example, you need STM32WL targets.
But QSPI is not supported at all for that STM32 family.

jeromecoutant, 2021-03-08:
See #14392

kvkc97, 2021-03-16:
I followed the above mentioned steps and successfully added QSPI but the classes and data types declared declared in the QSPI.h header file of mbed-os cannot be called in the application. I added the header file and declared an object of the QSPI class as below and it generated error.
#include "drivers/QSPI.h"
QSPI qspi_device(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN);
qspi_status_t result;
Both qspi_status_t and QSPI are declared in QSPI.h header file. The error obtained is attached.
[image]
Is there any step which  am missing in the implementation?

kvkc97, 2021-03-24:
Can I get an update on this query..

kvkc97, 2021-04-12:
Can I get an update on this query.. @0xc0170

0xc0170, 2021-04-13:
> See #14392

Did you use this PR as well? 

What is your mbed_config.h ? Does it contain `QSPI` ? What is QSPI_MX25R.h - not in the Mbed OS tree, is it? Does it include the config header file to get the definitions ?

kvkc97, 2021-04-13:
Yes I added the steps given in #14392 PR into the mbed_app.json file. The error still persists. I am integrating QSPI to the mbed lorawan example code. mbed_config.h file is not present in it. 
QSPI_MX25R.h is a custom added library included to QSPI operations. It has the header file #include "drivers/QSPI.h" in it. However, to test whether the issue is with this QSPI library, I added the #include "drivers/QSPI.h" in main.cpp and defined a QSPI object as below.
QSPI qspi_device(QSPI_FLASH1_IO0, QSPI_FLASH1_IO1, QSPI_FLASH1_IO2, QSPI_FLASH1_IO3, QSPI_FLASH1_SCK, QSPI_FLASH1_CSN);
It generates the same error.

kvkc97, 2021-04-19:
Can I have an update on this query.

0xc0170, 2021-04-19:
@kvkc97 would you be able to share simple code example that could help us reproducing an issue?

kvkc97, 2021-04-19:
I have used mbed-os-example-lorawan code to test from the given link.
https://github.com/ARMmbed/mbed-os-example-lorawan
To test QSPI, I added a custom target board for STM32WB55RGV6. Custom board addition is successful and code is compiled(without changing the example code). I modified the main.cpp file (changes at lines 57,62,63) and custom_targets.json file to add QSPI. I have attached the above mentioned files. I have also atached a zip folder of peripheral pins files also if needed.
[image]
[custom_targets.json.docx](https://github.com/ARMmbed/mbed-os/files/6336345/custom_targets.json.docx)
[mbed_app.json.docx](https://github.com/ARMmbed/mbed-os/files/6336365/mbed_app.json.docx)
[TARGET_MYTARGET.zip](https://github.com/ARMmbed/mbed-os/files/6336382/TARGET_MYTARGET.zip)

The error obtained is attached
[image]

0xc0170, 2022-07-22:
I am closing this issuequestion

closed 2022-07-22 by 0xc0170