ARMmbed/mbed-os #14398: Symbols unavailable on Mbed CLI2

ccli8, 2021-03-09:
### Description of defect

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.8.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
**Mbed CLI2**:: 7.2.1

#### How is this defect reproduced ? 
On Mbed CLI, the following symbols are generated and passed to compiler, linker, or both, but on Mbed CLI2, they are unavailable.

**BUILD/.../.profile.c**:
```
{
    "flags": [
        ......
        "-DMBED_RAM_SIZE=0x28000",
        "-DMBED_RAM_START=0x20000000",
        "-DMBED_ROM_SIZE=0x80000",
        "-DMBED_ROM_START=0x0",
        ......
```

**BUILD/.../.profile.ld**:
```
{
    "flags": [
        ......
        "--predefine=\"-DMBED_BOOT_STACK_SIZE=1024\"",
        "--predefine=\"-DMBED_RAM_SIZE=0x28000\"",
        "--predefine=\"-DMBED_RAM_START=0x20000000\"",
        "--predefine=\"-DMBED_ROM_SIZE=0x80000\"",
        "--predefine=\"-DMBED_ROM_START=0x0\"",
        ......
```

ccli8, 2021-03-09:
Updated issue template

rwalton-arm, 2021-03-09:
> I can't mention mbed-os-tools team (we will look at why it is not available anymore)
> 
> @rwalton-arm Should this be an issue in mbed-tools repository? If yes, I can move it there.

This looks the same as the following issue: https://github.com/ARMmbed/mbed-tools/issues/156 

It would be useful to add the comments here to that ticket. Thanks.

ccli8, 2021-03-10:
Re-create on [mbed-tools](https://github.com/ARMmbed/mbed-tools/issues/222). Close this one.

closed 2021-03-10 by ccli8