ARMmbed/mbed-os #14399: AnalogIn don't work on DISCO_L072CZ_LRWAN1

MathieuPossamai, 2021-03-09:
### Description of defect

I try to open an analog input on my DISCO_L072CZ_LRWAN1. When i do the card crash and return me the error :
   ```
 ++ MbedOS Error Info ++
    Error Status: 0x80FF0144 Code: 324 Module: 255
    Error Message: Assertion failed: obj->handle.Instance != (ADC_TypeDef *)NC
    Location: 0x80103CD
    File: /extras/mbed-os.lib/targets/TARGET_STM/TARGET_STM32L0/analogin_device.c+53
    Error Value: 0x0
    Current Thread: main Id: 0x20001D04 Entry: 0x8010D71 StackSize: 0x400 StackMem: 0x20001178 SP: 0x2000147C
```

#### Target(s) affected by this defect ?
DISCO_L072CZ_LRWAN1

#### What version of Mbed-os are you using (tag or sha) ?
MbedOS 6.8.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed online compiler

#### How is this defect reproduced ? 
Just create a new program with an AnalogIn command on an DISCO_L072CZ_LRWAN1 on MbedOS 6.8.0

0xc0170, 2021-03-09:
cc @ARMmbed/team-st-mcd

MathieuPossamai, 2021-03-09:
After more tries, i found out that only the pin A0 and the pin A2 are working. I think i can't use the other one as analog input...

jeromecoutant, 2021-03-09:
> So, the other pins are useless ?

Yes, these Arduino pins are not connected to MCU pins...

jeromecoutant, 2021-03-09:
See User Manual, it seems that you can connect 
A1 pins to PA_0 with SB7,
A3 to PA_4 with SB8,
A4 to PB_9 with SB11 or PH_0 with SB,
and A5 to PB_8 with SB12 or PH_1 with SB10

MathieuPossamai, 2021-03-09:
Ok, thanks for all !

closed 2021-03-09 by MathieuPossamai