ARMmbed/mbed-os #14404: Failure to build for GR_MANGO using CMake

Patater, 2021-03-10:
### Description of defect

```
Compiling example repos....

[EXAMPLES]> INFO     In folder 'mbed-os-example-blinky'
[EXAMPLES]> INFO     Compiling mbed-os-example-blinky GR_MANGO GCC_ARM
[EXAMPLES]> INFO     Executing command 'mbed-tools compile -t GCC_ARM -m GR_MANGO -c'...

#### STDOUT ####
Configuring project and generating build system...
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python3: /usr/bin/python3.7 (found version "3.7.4") found components: Interpreter 
-- Checking for Python package prettytable -- found
-- Checking for Python package future -- found
-- Checking for Python package jinja2 -- found
-- Checking for Python package intelhex -- found
-- Configuring incomplete, errors occurred!
See also "/builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@5/examples/mbed-os-example-blinky/cmake_build/GR_MANGO/develop/GCC_ARM/CMakeFiles/CMakeOutput.log".
See also "/builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@5/examples/mbed-os-example-blinky/cmake_build/GR_MANGO/develop/GCC_ARM/CMakeFiles/CMakeError.log".

#### STDERR ####
CMake Error at mbed-os/connectivity/drivers/emac/CMakeLists.txt:21 (add_subdirectory):
  The source directory

    /builds/workspace/mbed-os-ci_cmake-example-GCC_ARM-dev@5/examples/mbed-os-example-blinky/mbed-os/connectivity/drivers/emac/TARGET_RENESAS_EMAC

  does not contain a CMakeLists.txt file.

ERROR: CMake invocation failed!

More information may be available by using the command line option '-v'.

#### End of STDOUT/STDERR ####

[EXAMPLES]> INFO     In folder 'mbed-os-example-sockets'
[EXAMPLES]> INFO     In folder 'mbed-os-example-cellular'
[EXAMPLES]> INFO     In folder 'mbed-os-example-lorawan'
[EXAMPLES]> INFO     In folder 'mbed-os-example-devicekey'
[EXAMPLES]> INFO     In folder 'mbed-os-example-mbed-crypto/getting-started'
[EXAMPLES]> INFO     In folder 'mbed-os-example-nfc/NFC_EEPROM'
[EXAMPLES]> INFO     In folder 'mbed-os-example-nfc/NFC_SmartPoster'
[EXAMPLES]> INFO     In folder 'mbed-os-example-ble/BLE_Advertising'
[EXAMPLES]> INFO     In folder 'mbed-os-example-kvstore'

Passed Example Compilation:
+--------------+--------+-----------+----------+--------------+
| EXAMPLE NAME | TARGET | TOOLCHAIN | TEST GEN | BUILD RESULT |
+--------------+--------+-----------+----------+--------------+
+--------------+--------+-----------+----------+--------------+

Failed Example Compilation:
+------------------------+----------+-----------+----------+--------------+
| EXAMPLE NAME           |  TARGET  | TOOLCHAIN | TEST GEN | BUILD RESULT |
+------------------------+----------+-----------+----------+--------------+
| mbed-os-example-blinky | GR_MANGO |  GCC_ARM  |  UNSET   |    FAILED    |
+------------------------+----------+-----------+----------+--------------+
Number of failures = 1
```

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM (gcc-arm-none-eabi-9-2019-q4-major)

#### What version of Mbed-os are you using (tag or sha) ?

0e89f9d6d9ca

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools

#### How is this defect reproduced ? 

In folder 'mbed-os-example-blinky'
```
mbed-tools compile -t GCC_ARM -m GR_MANGO -c
```

Patater, 2021-03-10:
It appears that `RZ_A1H` is failing for the same root cause: `mbed-os/connectivity/drivers/emac/TARGET_RENESAS_EMAC does not contain a CMakeLists.txt file.`

0xc0170, 2021-03-11:
We will send a fix.

hugueskamba, 2021-03-11:
Hello,
I can build GR_MANGO and RZ_A1H using Mbed OS master and Blinky development .
I will raise a PR to remove the following warning:
```
[152/153] Building C object CMakeFiles/mbed-os-example-blinky.dir/mbed-os/targets/TARGET_RENESAS/TARGET_RZ_A2XX/r_can/src/r_can_rz.c.obj
cc1: warning: switch '-mcpu=cortex-a9' conflicts with '-march=armv7-a' switch
```

closed 2021-03-11 by 0xc0170