ARMmbed/mbed-os #14413: CMake netsockets: too many libraries dependencies

0xc0170, 2021-03-11:
### Description of defect

Build mbed-os-example-sockets with mbedtools (CMake) and check what files are being included. While we were looking at object libraries, we found these dependencies for netsockets:

```
target_link_libraries(mbed-netsocket
    PRIVATE
        mbed-nanostack
        mbed-randlib
        mbed-core
        mbed-rtos
        mbed-lwipstack
        mbed-nanostack-libservice
        mbed-ppp
    PUBLIC
        mbed-events
        mbed-mbedtls
)
```

This is also related to https://github.com/ARMmbed/mbed-os/issues/14412 (not certain if we clean up randlib, we would remove nanostack dependency for netsockets, we might then we should review if we need mbed-nanostack-libservice and mbed-ppp always. Questionable would be also mbedtls events as they are required only for secure sockets, or anything else in the tree?

#### Target(s) affected by this defect ?

all

#### Toolchain(s) (name and version) displaying this defect ?

all

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.8

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbedt-tools 7.0

#### How is this defect reproduced ? 

Build mbed-os-exaple-socket and explore build.ninja file what is include (number of files is a good indicator as well how much we are including for this example).

boraozgen, 2021-04-20:
Compiling 800+ files for an ethernet based socket example. Not much of an improvement on the old tools in this case :)

0xc0170, 2021-04-20:
> Compiling 800+ files for an ethernet based socket example. Not much of an improvement on the old tools in this case :)

Correct. I created these issues about deps so we can plan this work - most of us  are not aware of them as old tool hide this detail.  It's not trivial as it was just easy to add a header here, there -  building tree with "all in" for years.

It will require to go through the libraries and remove deps where they can.

ladislas, 2021-12-30:
Just hit this issue, compilation times are way to long because of this.

ladislas, 2021-12-31:
> Correct. I created these issues about deps so we can plan this work

@0xc0170 any ideas when things like this will be worked on? It seems like the CMake migration, though promising, has been lost in translation.