ARMmbed/mbed-os #14417: connectivity drivers include `mbed.h`

0xc0170, 2021-03-11:
### Description of defect

See the offenders below:

```
connectivity\drivers\802.15.4_RF\TARGET_Silicon_Labs\TARGET_SL_RAIL\NanostackRfPhyEfr32.cpp:
   18  #include <string.h>
   19  
   20: #include "mbed.h"
   21  #include "mbed_power_mgmt.h"
   22  #include "ns_types.h"

connectivity\drivers\802.15.4_RF\TARGET_Silicon_Labs\TARGET_SL_RAIL\NanostackRfPhyEfr32.h:
   18  #define NANOSTACK_PHY_EFR32_H_
   19  
   20: #include "mbed.h"
   21  #include "NanostackRfPhy.h"
   22  

connectivity\drivers\ble\FEATURE_BLE\TARGET_CY8C63XX\drivers\IPCPipeTransportDriver.h:
   22  
   23  #include <stdint.h>
   24: #include "mbed.h"
   25  #include "ble/driver/CordioHCITransportDriver.h"
   26  

connectivity\drivers\ble\FEATURE_BLE\TARGET_STM32WB\HCIDriver.cpp:
   21  #include "ble/driver/CordioHCIDriver.h"
   22  #include "ble/driver/CordioHCITransportDriver.h"
   23: #include "mbed.h"
   24  #include "hci_api.h"
   25  #include "hci_cmd.h"

connectivity\drivers\emac\TARGET_ARM_FM\COMPONENT_LAN91C111\fvp_emac.h:
   20  
   21  #include "EMAC.h"
   22: #include "mbed.h"
   23  #include "rtos/Mutex.h"
   24  #include "lan91c111.h"

connectivity\drivers\emac\TARGET_ARM_SSG\COMPONENT_SMSC9220\smsc9220_emac.h:
   20  
   21  #include "EMAC.h"
   22: #include "mbed.h"
   23  #include "rtos/Mutex.h"
   24  

connectivity\drivers\emac\TARGET_STM\stm32xx_emac.cpp:
   22  #include "cmsis_os.h"
   23  
   24: #include "mbed.h"
   25  #include "mbed_interface.h"
   26  #include "mbed_assert.h"

connectivity\drivers\wifi\TARGET_WICED\wiced_interface\WicedInterface.h:
   21  #if defined(MBED_CONF_NSAPI_PRESENT)
   22  
   23: #include "mbed.h"
   24  #include "EthernetInterface.h"
   25  #include "netsocket/OnboardNetworkStack.h"

connectivity\FEATURE_BLE\libraries\TARGET_CORDIO_LL\stack_adaptation\custom_chci_tr.cpp:
   15   */
   16  
   17: #include "mbed.h"
   18  #include "custom_chci_tr.h"
   19  #include "chci_tr.h"
```

I can create  this per each but it would lead to have too many issues to create so I grouped these to "connectivity drivers". This should be cleaned up if we want to better separate libraries.

#### Target(s) affected by this defect ?

See the list above

#### Toolchain(s) (name and version) displaying this defect ?

all

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.8

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Not reproducible

0xc0170, 2021-03-11:
I'll mention some of the teams that  could fix these.

@ARMmbed/team-st-mcd @ARMmbed/team-silabs @ARMmbed/team-cypress

jeromecoutant, 2021-03-11:
Hi @0xc0170 
I am checking
connectivity\drivers\ble\FEATURE_BLE\TARGET_STM32WB\HCIDriver.cpp and
connectivity\drivers\emac\TARGET_STM\stm32xx_emac.cpp