ARMmbed/mbed-os #14432: Compiler error for ST-Targets when stack dump is enabled

ghseb, 2021-03-16:
**Description of defect**
Compiler error due to undefined macro `INITIAL_SP` in
https://github.com/ARMmbed/mbed-os/blob/e4b81f67f939a0c0b11c147ce74aa367271e1279/platform/source/mbed_error.c#L537

The related change was introduced via https://github.com/ARMmbed/mbed-os/pull/12140/files.

**Target(s) affected by this defect ?**
STM32

**Toolchain(s) (name and version) displaying this defect ?**
n/a

**What version of Mbed-os are you using (tag or sha) ?**
https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.15.4

**What version(s) of tools are you using. List all that apply (E.g. mbed-cli)**
n/a

**How is this defect reproduced ?**
Set the following configuration parameters in `mbed_app.json`:
```
"platform.error-all-threads-info": true,
"platform.stack-dump-enabled": true,
```

jeromecoutant, 2021-03-16:
I think INITIAL_SP should be calculated from linker script ?

ghseb, 2021-03-16:
Yes generating it from the linker script would be best. Setting the macro via `mbed_app.json` or `targets.json` works but its ugly and error prone (as the previous solution with `mbed_rtx.h` was and actually is for non STM32 targets).

There should some API (or reuse `INITIAL_SP`) that allows to retrieve the value and internally uses compiler dependent functionality. E.g. the following for IAR:
```
#pragma segment="CSTACK"
__segment_end("CSTACK");
```

Patater, 2021-04-01:
This may be broken on any target, not just ST targets.

jeromecoutant, 2021-05-05:
@Patater #14622 is requesting the same correction...

0xc0170, 2021-05-05:
mbed_error uses `INITIAL_SP`. This is still documented as requirement (however it was removed from STM rtx file). See our docs: https://os.mbed.com/docs/mbed-os/v6.9/porting/porting-bootstrap.html

Steps to fix this:
- update mbed_error.c to use `__INITIAL_SP` that comes from cmsis (all toolchains)
- remove documentation requirement to defined `INITIAL_SP`

I'll send a fix now to do the first and we agree it is the right fix, I can fix docs as well.

0xc0170, 2021-05-05:
I created PR with the fix https://github.com/ARMmbed/mbed-os/pull/14624

PR #14624 (2021-05-11): error: use __INITIAL_SP from cmsis instead of RTX one

### Summary of changes 

We used to require INITIAL_SP as rtx target headers define it. This should not be required, as
cmsis already defines symbol  __INITIAL_SP for all toolchains.

Fixes #14432

We have `platform/tests/TESTS/mbed_platform/error_handling/main.cpp` but that does not probably contain any config in our CI to run with these enabled:

```
"platform.error-all-threads-info": true,
"platform.stack-dump-enabled": true,
```

#### Impact of changes
closed 2021-05-06 by ghseb