ARMmbed/mbed-os #14440: STM32L452RE: SPI1 + PA_6 digital IO = infinite hang

multiplemonomials, 2021-03-17:
### Description of defect

I have found that using both SPI1 and PA_6 (as a DigitalOut) on the STL32L452RE causes an infinite hang whenever code attempts to send a byte over the SPI bus.  I did make an effort to look into what's causing this, but wasn't able to find any smoking guns.  It looks like when PA_6 is in use, the STM SPI code loops infinitely checking `LL_SPI_IsActiveFlag_RXNE()`.  It's apparently supposed to set when received data is available in the RX FIFO, but it never sets.  Not sure what's going on, hopefully you guys can figure it out.

#### Target(s) affected by this defect ?

Custom board based on NUCLEO_L452RE_P

#### Toolchain(s) (name and version) displaying this defect ?

GCC 9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cmake beta 1.6.0

#### How is this defect reproduced ? 

I used the following code:

```cpp

#include <mbed.h>

int main()
{

	// Create SPI on SPI1
	SPI testSPI(PA_12, PA_11, PA_5);

	// Try an SPI transaction (no CS, so not actually communicating with anything)
	testSPI.write(0);
	printf("Transaction 1 complete!\n");

	// Create DIO on PA6
	DigitalOut dOut(PA_6, 1);

	testSPI.write(0);
	printf("Transaction 2 complete!\n");

	while(true) {
		printf("Done!\n");
		ThisThread::sleep_for(1s);
	}

	return 0;
}
```

If I comment out the DigitalOut or change it to a different pin like PA_7, I get:
```
Transaction 1 complete!
Transaction 2 complete!
Done!
```

But with the code as written, I get:
```
Transaction 1 complete!
```

0xc0170, 2021-03-18:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-03-18:
Hi
I am sorry but everything is working with my NUCLEO_L452RE_P target...
With 6.5.0 ans master branch.

LMESTM, 2021-03-18:
Could that indicate that there might be a short cut between PA_6 line and 1 of the SPI lines on your custom board ?

multiplemonomials, 2021-03-18:
Hmm, I just tested with Mbed Studio and Mbed 6.9 instead of my own project.  Same exact issue.  I wonder if it has something to do with me using the STM32L452RET6 instead of the chip version with SMPS?  Could that change the pin mappings in any way?

multiplemonomials, 2021-03-19:
Update: tried the same code on the board with the issue and another board with the exact same CPU.  As you suspected, it only breaks on that one board.  Not sure if it's the CPU or the soldering, but something's definitely up.  Sorry for wasting your time!

closed 2021-03-19 by multiplemonomials