ARMmbed/mbed-os #14446: CMake: Mbed is forcing toolchain from PATH

boraozgen, 2021-03-18:
Previously mentioned here: https://github.com/ARMmbed/mbed-tools/issues/153

The following lines force CMake to use the toolchain from PATH.

https://github.com/ARMmbed/mbed-os/blob/c73413893fb98aaaeda74513c981ac68adc8645d/tools/cmake/toolchains/GCC_ARM.cmake#L4-L7

This should instead be provided by the user (IDEs like VSCode and CLion do this). Otherwise it leads to configuration issues when using multiple toolchains in the same project. I have such a use case, where I build another target to run my GUI library on the host.

Is there a case against removing these lines? `GCC_ELF2BIN` could then be derived from the provided compiler location.

0xc0170, 2021-03-19:
We can use `find program` if not set. A user should supply this via -D to provide own paths.

> This should instead be provided by the user (IDEs like VSCode and CLion do this). Otherwise it leads to configuration issues when using multiple toolchains in the same project. I have such a use case, where I build another target to run my GUI library on the host.

Is it expected to do -DCMAKE_ASM_COMPILER=path ?

boraozgen, 2021-03-19:
I don't understand your proposition. Let me clarify my point.

In my opinion, `CMAKE_C_COMPILER` and `CMAKE_CXX_COMPILER` variables should be set by the IDE or by the user with `-D`, instead of by Mbed.

What do you mean by `find program`? Do you mean the `find_program` function of CMake? What is the purpose of using it?

ladislas, 2021-03-19:
The user can also provide those in their main CMakeLists.txt. If not provided, a warning could be emitted and the default could be set. This way it works as before but lets user customize those.

I'm not sure about the `find_program` either for something as important as the compiler. For ccache or cppcheck and other tools that are not mandatory, it's okay, but you must provide a compiler for this to work.

> I have such a use case, where I build another target to run my GUI library on the host.

@boraozgen can you tell me more about that? what IDE do you use? How did you set it up? With VSCode and CMake Tools, it's not possible to have two toolchains live side by side for the same project :(

boraozgen, 2021-03-19:
I use vscode + cmake-tools. cmake-tools has the option to change the toolchains (called kits in the tool). It currently does not work with Mbed as it overrides this variable.

0xc0170, 2021-03-19:
> In my opinion, CMAKE_C_COMPILER and CMAKE_CXX_COMPILER variables should be set by the IDE or by the command line with -D, instead of by Mbed.

👍 I've got what I was looking for. We will triage and fix. it should be provided via -D, not being in CMakelists.txt

boraozgen, 2021-03-19:
This also might apply to other toolchain related stuff like compiler options. How about a possibility to make the user supply an optional toolchain file, and using the default GCC_ARM file by default?

0xc0170, 2021-03-22:
> This also might apply to other toolchain related stuff like compiler options. How about a possibility for the user to supply an optional toolchain file, and using the current GCC_ARM file by default?

That is the idea. if `CMAKE_TOOLCHAIN_FILE` is not provided, we supply the default Mbed one.

boraozgen, 2021-03-22:
Exactly. Apparently that conditional is already there:

https://github.com/ARMmbed/mbed-os/blob/96e19afdd196c6c99edd58fddd44e2c691cdca2f/tools/cmake/app.cmake#L12-L16

There are two issues with the current state:
1. `toolchain.cmake` uses variables like `MBED_PATH` and `MBED_TOOLCHAIN`, which are not defined when toolchain files are processed. Therefore it is not possible to provide the default `toolchain.cmake` as `CMAKE_TOOLCHAIN_FILE`.
2. `toolchain.cmake` includes both Mbed-specific and user-specific settings. These should be separated in order to avoid adding too much boilerplate code to the user-specified toolchain file.

0xc0170, 2021-03-22:
https://github.com/ARMmbed/mbed-os/issues/14385 - MBED_PATH removal issue

Can you create an issue for 2nd?

boraozgen, 2021-03-22:
I edited the issue title to reflect the updated info. #14385 refers to only `MBED_PATH` but there are other constructs in `toolchain.cmake` which don't work as a real toolchain file. I think the goal here is to be able to provide a minimal custom toolchain file without the need to copy all of the functions and Mbed related settings etc. from `toolchain.cmake`. We should define the exact requirements, such as compiler flags, etc. I am not very experienced with toolchain files so maybe someone could help us out here.

0xc0170, 2021-03-22:
OK, I'll link here the proposal from Rob: https://github.com/ARMmbed/mbed-tools/discussions/225 (this should be tracking issue for section "Improve toolchain handling"