ARMmbed/mbed-os #14448: Cellular Error

symeonmattes, 2021-03-18:
### Description of defect

#### Target(s) affected by this defect ?
[Embedded Planet Agora](https://os.mbed.com/platforms/AGORA-DEV/) 

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os (#26606218ad9d, tags: mbed-os-6.8.0, mbed-os-6.8.0-rc1)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli: 1.10.4
Mbed Studio: 1.3.1

#### How is this defect reproduced ? 

I run the project [mbed-os-example-cellular](https://github.com/ARMmbed/mbed-os-example-cellular) and I setup the [mbed_app.json](https://github.com/ARMmbed/mbed-os-example-cellular/blob/master/mbed_app.json) changing only the values:

```
  "cellular.debug-at": true,
  "nsapi.default-cellular-sim-pin": "\"1234\"",
  "nsapi.default-cellular-plmn": null,
  "nsapi.default-cellular-apn": "\"internet\"",
  "nsapi.default-cellular-username": "\"user\"",
  "nsapi.default-cellular-password": "\"user\""
```
I have tried the sim card in a mobile device and it connects properly to 3G and I have coppied all the configuration from there. So I suppose the sim card is ok. However when I run the code I receive the following logs:

```
mbed-os-example-cellular
[00000000ms][DBG ][CELL]: Modem flow control: RTS 14 CTS 40
[00000002ms][INFO][CELL]: New CellularContext  (0x2000AC60)
Establishing connection
[00000003ms][INFO][CELL]: CellularContext connect
[00000012ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00000092ms][INFO][CELL]: Start connecting (timeout 1000 ms)
[00000171ms][INFO][CELL]: AT TX ( 2): AT
[00000230ms][INFO][CELL]: AT TX ( 1): <cr>
[00001172ms][DBG ][CELL]: AT error -3012
[00001172ms][INFO][CELL]: AT TX ( 2): AT
[00001173ms][INFO][CELL]: AT TX ( 1): <cr>
[00002173ms][DBG ][CELL]: AT error -3012
[00002173ms][DBG ][CELL]: Init => Power
[00002174ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00002175ms][INFO][CELL]: Modem power ON (timeout 1000 ms)
[00002178ms][DBG ][CELL]: Power => Device ready
[00002244ms][DBG ][CELL]: callback: 4106, err: 0, data: 4096
[00002323ms][DBG ][CELL]: Device was not ready, calling soft_power_on()
[00027614ms][DBG ][CELL]: AT flush
[00027615ms][INFO][CELL]: AT RX (32): AT<cr>AT<cr><cr><ln>OK<cr><ln><cr><ln>OK<cr><ln><cr><ln>#STN: 33,00,
[00027616ms][INFO][CELL]: AT RX (32): "Welcome to Vodafone"<cr><ln><cr><ln>#QSS: 1
[00027629ms][INFO][CELL]: AT RX ( 2): <cr><ln>
[00027693ms][INFO][CELL]: AT TX ( 4): ATE0
[00027754ms][INFO][CELL]: AT TX ( 1): <cr>
[00027815ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00027910ms][INFO][CELL]: AT TX ( 9): AT+CMEE=1
[00027956ms][INFO][CELL]: AT TX ( 1): <cr>
[00028016ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00028111ms][INFO][CELL]: AT TX ( 9): AT+CFUN=1
[00028157ms][INFO][CELL]: AT TX ( 1): <cr>
[00028218ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00028313ms][INFO][CELL]: AT TX (13): AT&K3;&C1;&D0
[00028363ms][INFO][CELL]: AT TX ( 1): <cr>
[00028424ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00028519ms][INFO][CELL]: AT TX ( 8): AT#QSS=1
[00028564ms][INFO][CELL]: AT TX ( 1): <cr>
[00028624ms][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00028719ms][INFO][CELL]: AT TX ( 9): AT#PSNT=1
[... 304 lines cut ...]
[00924205ms][DBG ][CELL]: callback: 4104, err: 0, data: 4098
[00924284ms][INFO][CELL]: RSSI unknown
[00924341ms][DBG ][CELL]: callback: 4106, err: 0, data: 4098
[00924420ms][INFO][CELL]: AT TX ( 9): AT+CEREG?
[00924486ms][INFO][CELL]: AT TX ( 1): <cr>
[00924547ms][INFO][CELL]: AT RX (21): <cr><ln>+CEREG: 0,0<cr><ln><cr><ln>OK<cr><ln>
[00924650ms][DBG ][CELL]: +CEREG: NotRegistered, LAC -1, cell -1, RAT unknown
[00924768ms][INFO][CELL]: AT TX ( 9): AT+CGREG?
[00924814ms][INFO][CELL]: AT TX ( 1): <cr>
[00924874ms][INFO][CELL]: AT RX (21): <cr><ln>+CGREG: 0,0<cr><ln><cr><ln>OK<cr><ln>
[00924978ms][DBG ][CELL]: +CGREG: NotRegistered, LAC -1, cell -1, RAT unknown
[00925075ms][INFO][CELL]: Network registration (timeout 180000 ms)
[00925161ms][DBG ][CELL]: callback: 4106, err: 0, data: 4098
[00925240ms][DBG ][CELL]: Registering network: retry 9/10
[00925317ms][DBG ][CELL]: callback: CellularStateRetryEvent, err: 0, data: 4098, retrycount: 9
[00925432ms][INFO][CELL]: Continue after 1200 seconds
[01800012ms][WARN][CELL]: No cellular connection
Couldn't connect: -3019, will retry
[01800013ms][INFO][CELL]: CellularContext connect
[01800014ms][WARN][CELL]: Canceling ongoing event (12288)

```
I was checking around to check the error code numbers but not sure what they mean.

Any suggestions?

0xc0170, 2021-03-19:
cc @ARMmbed/team-embeddedplanet

symeonmattes, 2021-03-24:
@ciarmcom Could you please let me know if you have anything?

symeonmattes, 2021-03-26:
Based on [Telit ME910C1 AT Commands Reference Guide](https://nimbelink.com/Documentation/Skywire/4G_LTE_Cat_M1_Telit/1001623_TM1G_AT_Command_Manual.pdf) "AT error code 11" means "SIM PIN required".

I removed the pin code from the sim card so as not to require any authentication, but I still can't connect. Not sure what "AT error -3012" means