ARMmbed/mbed-os #14452: STM32 EMAC does not support EthernetInterface.set_mac_address

JojoS62, 2021-03-19:
### Description of defect

calling set_mac_address() for STM32 targets returns NSAPI_ERROR_UNSUPPORTED, so it is not implemented.
The HAL supports setting a MAC address by ETH_MACAddressConfig().
Was there a reason for not implementing this function or are the drivers not ready for this? 
Is there a schedule for adding this function? Otherwise I would try to make a PR when I get it working.
Setting the MAC is important for custom boards to companies MAC ranges.

#### Target(s) affected by this defect ?

STM32, tested with Nucleo STM32F46ZG

#### Toolchain(s) (name and version) displaying this defect ?

gcc 9.x

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli

#### How is this defect reproduced ? 

call eth.set_mac_address(myMAC, sizeof(myMAC); before net.connect();

jeromecoutant, 2021-03-19:
> Was there a reason for not implementing this function or are the drivers not ready for this?

No...

> Is there a schedule for adding this function?

I am afraid not...

JojoS62, 2021-03-19:
ok, thanks for your fast answer. In theory, it should be easy by using the HAL function, I'll give it a try.

katherrafi, 2021-06-17:
To change the default MAC address in STM32, 
If we have the function mbed_otp_mac_address() in the user application like below, 
the ethernet address can be changed via manually.  Because as this is defined as weak  in stm32xx_emac.cpp
___weak_ uint8_t mbed_otp_mac_address(char *mac).

The below code is responsible for the MAC address assignement.
void mbed_mac_address(char *mac)
{
      if (mbed_otp_mac_address(mac)) {         **_// Hence if we have user function here, we can able to assign the new MAC._**
        return;
    } else {
        mbed_default_mac_address(mac);
    }
    return;
}
This is the method designed in the existing HAL and its working well.

Please find the code snippet here

`..`
`uint8_t mbed_otp_mac_address(char *mac);`
`uint8_t mbed_otp_mac_address(char *mac)`                                           
`{`                                                                                 
`    unsigned char ST_mac_addr[6] = {0x00, 0x88, 0xe0,0x90,0x80,0x70}; // New User mac address`
`//    printf("%s:%s\n",__FILE__,__func__);`                                        
`    memcpy(mac,ST_mac_addr,sizeof(ST_mac_addr));`                                  
`    return 1;`                                                                     
`}`                                                                                 
                                                                                    
                                                                                    
`int main()`                                                                        
`{`                                                                                 
`    // Bring up the ethernet interface`                                            
`    printf("Ethernet socket example\n");`                                          
`    uint8_t MyMAC[6];`                                                             
`    printf("return of set_mac_address:%d\n",net.set_mac_address(MyMAC,sizeof(MyMAC)));`  
                                                                                   
`    net.connect();`                                                                
`    printf("MAC address %s\n",net.get_mac_address());` 
`...`

Output as follows:
**Ethernet socket example
return of set_mac_address:0
MAC address 00:88:e0:90:80:70**
IP address: 192.168.1.7
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]
Done

JojoS62, 2021-06-17:
thanks, that will be fine as a workaround. set_mac_address() should work also, but this has a low priority for me at the moment.

jeromecoutant, 2021-06-22:
> Setting the MAC is important for custom boards to companies MAC ranges.

I think this mbed_otp_mac_address function can be considered as the solution ?

Maybe few lines can be added in https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM#ethernet-configuration?

jeromecoutant, 2021-07-08:
Could we close this one with #14838 ?
Thx

JojoS62, 2021-07-08:
Yes, thanks.

closed 2021-07-08 by JojoS62