ARMmbed/mbed-os #14468: USBKeyboard.cpp not compiled and linked on FRDM-K22F

Timmmm, 2021-03-25:
### Description of defect

When you add the following code to the Mbed OS 6 bare metal blinky example:

```
#include "USBKeyboard.h"

USBKeyboard key;
```

And compile for FRDM-K22F then it doesn't compile `USBKeyboard.cpp` and consequently you get a link error like this:

```
[Error] @0,0: L6218E: Undefined symbol USBKeyboard::USBKeyboard(bool, unsigned short, unsigned short, unsigned short) (referred from BUILD/K22F/ARMC6/main.o).
```

#### Target(s) affected by this defect ?

At least FRDM-K22F. I haven't checked others.

#### Toolchain(s) (name and version) displaying this defect ?

Not sure what toolchain means in this context? I just downloaded the current version of Mbed Studio and used that.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

By the way the instructions in the issue template are out of date. The file is at `/mbed-os/platform/include/platform/mbed_version.h`, not `/platform/mbed_version.h`.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.3.1
`mbed --version` 1.10.4

#### How is this defect reproduced ? 

See above.

By the way the example on [this page](https://os.mbed.com/docs/mbed-os/v6.9/apis/usbkeyboard.html) for `USBKeyboard` appears to be wrong. It has no `printf()` method.

Timmmm, 2021-03-25:
Oh, I just noticed that someone posted a solution to this on the forum topic I made [here](https://forums.mbed.com/t/usbkeyboard-undefined-when-compiling-for-k22f/12617/2):

> if you have in your mbed_app.json file only this content
>
>     {
>         "requires": ["bare-metal"]
>     }
>
> then you probably need to change it to
>
>    {
>         "requires": ["bare-metal", "events", "drivers-usb"]
>     }

That worked! It would be nice if it was documented somewhere though - ideally in the API docs for all the USB classes, which is probably where most people are going to go (rather than in some part of a guide that people aren't going to see).

Also it would be really cool if you had some kind of scanner of the compilation output that suggested user-friendly solutions for stuff like this. I may know what a link error is but someone just starting out isn't going to have a clue.

You could easily add a regex for `Undefined symbol USBKeyboard::USBKeyboard` and pop up a notification "It looks like you're having USB compilation errors. See [this page] to find out how to fix them", or something like that.

Btw I hope this isn't too critical; Mbed Studio is a *vast* *vast* improvement over all of your previous solutions (yotta, the online compiler, etc.) - it's getting really close to the point where it is as idiot-proof as Arduino, but with an enormously better API and IDE.

closed 2021-03-25 by Timmmm

0xc0170, 2021-04-07:
>That worked! It would be nice if it was documented somewhere though - ideally in the API docs for all the USB classes, which is probably where most people are going to go (rather than in some part of a guide that people aren't going to see).

This should be addressed by cli 2 where you would link to mbed-baremetal, mbed-events and mbed-usb libraries via target_link_libraries in your app.

As the old tools were frozen, we rather focus on the new tooling and make it better.

> Btw I hope this isn't too critical; Mbed Studio is a vast vast improvement over all of your previous solutions (yotta, the online compiler, etc.) - it's getting really close to the point where it is as idiot-proof as Arduino, but with an enormously better API and IDE.

Thanks for the feedback