ARMmbed/mbed-os #14470: USBKeyboard does not work with FRDM-K22F

Timmmm, 2021-03-25:
### Description of defect

I compiled and deployed the following program for FRDM-K22F

```
#include "mbed.h"
#include "USBKeyboard.h"

USBKeyboard key;

#define WAIT_TIME_MS 500 
DigitalOut led1(LED1);

int main()
{
    while (true)
    {
        led1 = !led1;
        thread_sleep_for(WAIT_TIME_MS);
    }
}
```

After applying [this fix](https://github.com/ARMmbed/mbed-os/issues/14468#issuecomment-806514383) it compiles, but when deployed to the board the board seems to freeze. The LED does not blink, and if I plug in the USB device into a Windows machine then Windows says the USB device has malfunctioned.

If I comment out `USBKeyboard key;` the LED blinks.

#### Target(s) affected by this defect ?

FRDM-K22F

#### Toolchain(s) (name and version) displaying this defect ?

Not sure what toolchain means in this context? I just downloaded the current version of Mbed Studio and used that.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.3.1
`mbed --version` 1.10.4

#### How is this defect reproduced ? 

See above.

Timmmm, 2021-03-25:
Device manager has this to say:

[image]

hugueskamba, 2021-03-25:
Can you please explain why `key` is declared but do not used?

Timmmm, 2021-03-25:
Just because I was making a minimal example. By the way it works with Mbed OS 2 using the online compiler (and changing `thread_sleep_for()` to `wait(0.5)`).

0xc0170, 2021-04-07:
cc @ARMmbed/team-nxp