ARMmbed/mbed-os #14475: STM32L1: ADC after deep sleep

chrissnow, 2021-03-25:
### Description of defect
After deep sleeping the ADC will not function as the HSI clock is no longer enabled.

We are using the following to bring the ADC back to life
```
void forceAdcOutOfSleep() {
    RCC_OscInitTypeDef HSIRCC_OscInitStruct;
    HAL_RCC_GetOscConfig(&HSIRCC_OscInitStruct);
    if (HSIRCC_OscInitStruct.HSIState != RCC_HSI_ON) {
        HSIRCC_OscInitStruct.HSIState = RCC_HSI_ON;
        HSIRCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
        HSIRCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
        HAL_StatusTypeDef ret = HAL_RCC_OscConfig(&HSIRCC_OscInitStruct);
        if (ret != HAL_OK) {
            debug("HSI initialization failed - ADC will not function properly\r\n");
        }
    }
}
```

#### Target(s) affected by this defect ?
XDOT_L151CC

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
Enable Tickless, though it may be possible to call deep sleep directly?
You should get a value the first time, then zero afterwards.

```
#include "mbed.h"

// Blinking rate in milliseconds
#define BLINKING_RATE     500ms

int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(LED1);
	AnalogIn   ain(A0);
	AnalogIn   vref(ADC_VREF);

    while (true) {
        led = !led;

        printf("vref: 0x%04X \n", vref.read_u16());
        ThisThread::sleep_for(BLINKING_RATE);
    }
}
```

chrissnow, 2021-03-25:
One for @jeromecoutant I think.

jeromecoutant, 2021-03-26:
I remind that we don't recommend to enable TICKLESS for STM32L1...

chrissnow, 2021-03-26:
> I remind that we don't recommend to enable TICKLESS for STM32L1...

I did somewhat expect that response! any particular reason on the L1? seems to be mostly behaving itself for us.

jeromecoutant, 2021-03-29:
> > I remind that we don't recommend to enable TICKLESS for STM32L1...
> 
> I did somewhat expect that response! any particular reason on the L1? seems to be mostly behaving itself for us.

Depending on your use case, we don't ensure a correct behaviour for targets that enables LPTICKER on top of ST RTC wakeup timer.
In other words, we let you decide and verify for your application.
Maybe you could have to update "lpticker_delay_ticks" value:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L1237-L1240

PR #14478 (2021-04-07): STM32L1 : enable ADC after deepsleep
### Summary of changes 

See STM32L1 Reference Manual:
12.3.2 ADC clock
When entering Stop mode, the ADC analog and digital interfaces remain inactive as the HSI and PCLK2 are disabled. Since the HSI is still deactived after resuming from Stop mode, the user must enable the HSI as the ADC analog interface clock source and continue using ADC conversions.

Fix #14475 

@chrissnow 

#### Impact of changes 

#### Migration actions required 

### Documentation 

-------------------------------------------------------------------------------------
closed 2021-04-07 by 0xc0170