ARMmbed/mbed-os #14476: DISCO_L562QE - device_name missing in targets.json

Sandra11823, 2021-03-26:
### Description of defect

 If you build a program with the DISCO_L562QE as target, you will get the following message:
```
Configuration error: Bootloader not supported on this target: targets.json `device_name` not specified.
```
This menas, that projects with a bootloader are not possible with this board at the moment.

I suggest, that the corresponding name from the CMSIS-Pack must be added to `targets.json`.

#### Target(s) affected by this defect ?

DISCO_L562QE

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6 (version 6.14)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os 6.4.0 (8ef0a435)
mbed-os 6.9.0 (c734138)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed Studio 1.3.1

#### How is this defect reproduced ? 

Build a project with the target DISCO_L562QE.

jeromecoutant, 2021-03-26:
Hi
I agree STM32L5 targets are missing in the arm_pack_manager index
I am checking it now

jeromecoutant, 2021-03-26:
@Sandra11823 could you check #14477 ?
Thx

Sandra11823, 2021-03-26:
I have created a test project (mbed-os-empty), added your changes and it built without the configuration error. 
I will integrate your changes into my project next week.

Thanks

PR #14477 (2021-03-31): STM32: Add STM32L5 in arm_pack_manager index file
### Summary of changes 

This fixes #14476 

STM32L5 targets were missing in the arm_pack_manager index.

targets.json file is updated accordingly.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2021-03-31 by adbridge