ARMmbed/mbed-os #14484: Standard Pin Names Migration

gpsimenos, 2021-03-29:
### Description of defect

We have recently introduced new pin name standards in Mbed. Refer to the [Standard Pin Names Porting Guide](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/standard_pin_names/pin_names_porting.md) for details. There are several targets that are not yet compliant with the new standard, as reported by the pinvalidate tool.

#### Generic pin names

| Target                   | Result     | Errors |
|--------------------------|------------|--------|
|  ARCH_MAX                |  FAILED    | 1      |
|  ARCH_PRO                |  FAILED    | 1      |
|  ARDUINO_NANO33BLE       |  FAILED    | 1      |
|  ARM_CM3DS_MPS2          |  FAILED    | 1      |
|  ARM_MPS2_M0             |  FAILED    | 1      |
|  ARM_MPS2_M0P            |  FAILED    | 1      |
|  ARM_MPS2_M3             |  FAILED    | 1      |
|  ARM_MPS2_M4             |  FAILED    | 1      |
|  ARM_MPS2_M7             |  FAILED    | 1      |
|  ARM_MUSCA_B1            |  FAILED    | 2      |
|  ARM_MUSCA_S1            |  FAILED    | 2      |
|  CY8CKIT_062_BLE         |  FAILED    | 1      |
|  CY8CKIT_062_WIFI_BT     |  FAILED    | 1      |
|  CY8CKIT_062S2_43012     |  FAILED    | 1      |
|  CY8CKIT064B0S2_4343W    |  FAILED    | 1      |
|  CY8CPROTO_062_4343W     |  FAILED    | 1      |
|  CY8CPROTO_062S3_4343W   |  FAILED    | 1      |
|  CYSBSYSKIT_01           |  FAILED    | 1      |
|  CYTFM_064B0S2_4343W     |  FAILED    | 1      |
|  CYW9P62S1_43012EVB_01   |  FAILED    | 1      |
|  CYW9P62S1_43438EVB_01   |  FAILED    | 1      |
|  DISCO_F413ZH            |  FAILED    | 2      |
|  DISCO_F429ZI            |  FAILED    | 2      |
|  DISCO_F469NI            |  FAILED    | 1      |
|  DISCO_F746NG            |  FAILED    | 2      |
|  DISCO_F769NI            |  FAILED    | 2      |
|  DISCO_L072CZ_LRWAN1     |  FAILED    | 1      |
|  EFM32GG_STK3700         |  FAILED    | 2      |
|  EFM32GG11_STK3701       |  FAILED    | 2      |
|  EP_AGORA                |  FAILED    | 2      |
|  EP_ATLAS                |  FAILED    | 1      |
|  EV_COG_AD3029LZ         |  FAILED    | 2      |
|  EV_COG_AD4050LZ         |  FAILED    | 2      |
|  FF_LPC546XX             |  FAILED    | 1      |
|  FF1705_L151CC           |  FAILED    | 2      |
|  FVP_MPS2_M0             |  FAILED    | 1      |
|  FVP_MPS2_M0P            |  FAILED    | 1      |
|  FVP_MPS2_M3             |  FAILED    | 1      |
|  FVP_MPS2_M4             |  FAILED    | 1      |
|  FVP_MPS2_M7             |  FAILED    | 1      |
|  GD32_F307VG             |  FAILED    | 1      |
|  GD32_F450ZI             |  FAILED    | 1      |
|  GR_LYCHEE               |  FAILED    | 1      |
|  GR_MANGO                |  FAILED    | 1      |
|  HEXIWEAR                |  FAILED    | 2      |
|  K22F                    |  FAILED    | 2      |
|  K64F                    |  FAILED    | 2      |
|  K66F                    |  FAILED    | 2      |
|  K82F                    |  FAILED    | 2      |
|  KL25Z                   |  FAILED    | 2      |
|  KL43Z                   |  FAILED    | 2      |
|  KL46Z                   |  FAILED    | 2      |
|  KW41Z                   |  FAILED    | 2      |
|  LPC1114                 |  FAILED    | 2      |
|  LPC1768                 |  FAILED    | 1      |
|  LPC54114                |  FAILED    | 2      |
|  LPC546XX                |  FAILED    | 2      |
|  MAX32620FTHR            |  FAILED    | 2      |
|  MAX32625MBED            |  FAILED    | 1      |
|  MAX32625PICO            |  FAILED    | 2      |
|  MAX32630FTHR            |  FAILED    | 2      |
|  MIMXRT1050_EVK          |  FAILED    | 2      |
|  MOTE_L152RC             |  FAILED    | 2      |
|  MTS_DRAGONFLY_F411RE    |  FAILED    | 2      |
|  MTS_DRAGONFLY_F413RH    |  FAILED    | 3      |
|  MTS_MDOT_F411RE         |  FAILED    | 2      |
|  NRF52_DK                |  FAILED    | 1      |
|  NRF52840_DK             |  FAILED    | 1      |
|  NUCLEO_F070RB           |  FAILED    | 2      |
|  NUCLEO_F072RB           |  FAILED    | 2      |
|  NUCLEO_F091RC           |  FAILED    | 2      |
|  NUCLEO_F103RB           |  FAILED    | 2      |
|  NUCLEO_F207ZG           |  FAILED    | 2      |
|  NUCLEO_F303K8           |  FAILED    | 2      |
|  NUCLEO_F303RE           |  FAILED    | 2      |
|  NUCLEO_F303ZE           |  FAILED    | 2      |
|  NUCLEO_F401RE           |  FAILED    | 2      |
|  NUCLEO_F411RE           |  FAILED    | 2      |
|  NUCLEO_F412ZG           |  FAILED    | 2      |
|  NUCLEO_F413ZH           |  FAILED    | 2      |
|  NUCLEO_F429ZI           |  FAILED    | 2      |
|  NUCLEO_F439ZI           |  FAILED    | 2      |
|  NUCLEO_F446RE           |  FAILED    | 2      |
|  NUCLEO_F446ZE           |  FAILED    | 2      |
|  NUCLEO_F746ZG           |  FAILED    | 2      |
|  NUCLEO_F756ZG           |  FAILED    | 2      |
|  NUCLEO_F767ZI           |  FAILED    | 2      |
|  NUCLEO_G031K8           |  FAILED    | 1      |
|  NUCLEO_G071RB           |  FAILED    | 2      |
|  NUCLEO_G431KB           |  FAILED    | 1      |
|  NUCLEO_G474RE           |  FAILED    | 2      |
|  NUCLEO_L073RZ           |  FAILED    | 2      |
|  NUCLEO_L152RE           |  FAILED    | 2      |
|  NUCLEO_WB55RG           |  FAILED    | 2      |
|  NUCLEO_WL55JC           |  FAILED    | 1      |
|  NUMAKER_IOT_M252        |  FAILED    | 3      |
|  NUMAKER_IOT_M263A       |  FAILED    | 3      |
|  NUMAKER_IOT_M487        |  FAILED    | 3      |
|  NUMAKER_PFM_M453        |  FAILED    | 2      |
|  NUMAKER_PFM_M487        |  FAILED    | 3      |
|  NUMAKER_PFM_NANO130     |  FAILED    | 2      |
|  NUMAKER_PFM_NUC472      |  FAILED    | 2      |
|  PORTENTA_H7_M4          |  FAILED    | 1      |
|  PORTENTA_H7_M7          |  FAILED    | 1      |
|  RZ_A1H                  |  FAILED    | 1      |
|  S1SBP6A                 |  FAILED    | 2      |
|  S5JS100                 |  FAILED    | 2      |
|  SDP_K1                  |  FAILED    | 1      |
|  SDT32620B               |  FAILED    | 2      |
|  SDT32625B               |  FAILED    | 2      |
|  SDT52832B               |  FAILED    | 1      |
|  SDT64B                  |  FAILED    | 1      |
|  SFE_ARTEMIS             |  FAILED    | 1      |
|  SFE_ARTEMIS_ATP         |  FAILED    | 1      |
|  SFE_ARTEMIS_DK          |  FAILED    | 1      |
|  SFE_ARTEMIS_MODULE      |  FAILED    | 1      |
|  SFE_ARTEMIS_NANO        |  FAILED    | 1      |
|  SFE_ARTEMIS_THING_PLUS  |  FAILED    | 1      |
|  SFE_EDGE                |  FAILED    | 1      |
|  SFE_EDGE2               |  FAILED    | 1      |
|  TB_SENSE_12             |  FAILED    | 2      |
|  TMPM46B                 |  FAILED    | 1      |
|  TMPM4G9                 |  FAILED    | 1      |
|  UHURU_RAVEN             |  FAILED    | 2      |
|  WIO_3G                  |  FAILED    | 2      |
|  WIO_BG96                |  FAILED    | 2      |
|  WIO_EMW3166             |  FAILED    | 2      |
|  XDOT_L151CC             |  FAILED    | 2      |

#### Target(s) affected by this defect ?
All targets in above table.

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
`python hal/tests/pinvalidate/pinvalidate.py -a -v`

notronrj, 2021-05-05:
I've submitted a PR that fixes MIMXRT1050_EVK.

LDong-Arm, 2021-06-28:
I've created #14837 to fix K64F