ARMmbed/mbed-os #14493: `std::chrono::system_clock::now()` does not update when `set_time` is called

AGlass0fMilk, 2021-03-31:
### Description of defect

A program similar to the following will not display the expected result:

```
#include "mbed.h"
#include "mbed_rtc_time.h"
#include <chrono>

using namespace std::chrono;

int main() {
   set_time(1609459200); // set time to 1/1/2021 midnight
   while(true) {
      printf("current epoch time: %d", duration_cast<milliseconds>(system_clock::now().time_since_epoch()).count());
      rtos::ThisThread::sleep_for(1s);
   }
}
```

#### Target(s) affected by this defect ?

Presumably all. Tested on NRF52840_DK with RTC enabled. (ie: device_has_add: ["RTC"])

#### Toolchain(s) (name and version) displaying this defect ?

```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

#### What version of Mbed-os are you using (tag or sha) ?

8340ea2d2b39d12a5220928b30c4b3dc90fd5dff

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
Package             Version     Location
------------------- ----------- -------------------------------------------------------------------
appdirs             1.4.3
asn1ate             0.6.0
attrs               19.3.0
Automat             20.2.0
beautifulsoup4      4.6.3
ble-serial          1.2.1
bleak               0.9.1
bluepy              1.3.0
cbor                1.0.0
certifi             2019.11.28
cffi                1.14.1
chardet             3.0.4
click               7.1
cmsis-pack-manager  0.2.10
cobs                1.1.4
cogapp              3.0.0
colorama            0.3.9
constantly          15.1.0
crc16               0.1.1
crccheck            0.6
cryptography        2.9.2
cycler              0.10.0
dataclasses         0.8
distlib             0.3.1
docopt              0.6.2
ecdsa               0.15
elftools            0.1.0.dev0
fasteners           0.15
filelock            3.0.12
Flask               1.1.2
future              0.16.0
futures             3.1.1
fuzzywuzzy          0.18.0
gitdb               4.0.5
GitPython           3.1.13
grip                4.5.2
hidapi              0.9.0.post2
hyperlink           20.0.1
icetea              1.2.4
idna                2.7
imgtool             1.7.0rc1
importlib-metadata  1.6.0
importlib-resources 5.1.2
incremental         17.5.0
inflection          0.5.1
iniconfig           1.1.1
intelhex            2.2.1
itsdangerous        1.1.0
Jinja2              2.10.3
jsonmerge           1.7.0
jsonschema          2.6.0
junit-xml           1.8
keyrings.alt        4.0.2
kiwisolver          1.2.0
lockfile            0.12.2
Logbook             1.5.3
Mako                1.1.4
manifest-tool       1.5.2
Markdown            3.3.2
MarkupSafe          1.1.1
matplotlib          3.3.0
mbed-cli            1.10.4
mbed-cloud-sdk      2.0.8
mbed-flasher        0.10.1
mbed-greentea       1.7.4
mbed-host-tests     1.5.10
mbed-ls             1.7.12
mbed-os-tools       0.0.15
mbed-tools          7.1.2
milksnake           0.1.5
monotonic           1.5
numpy               1.19.1
packaging           20.4
path-and-address    2.0.1
pc-ble-driver-py    0.14.2
pdoc3               0.9.2
Pillow              7.2.0
pip                 20.2
pkg-resources       0.0.0
pluggy              0.13.1
prettytable         0.7.2
protobuf            3.5.2.post1
psutil              5.6.6
py                  1.9.0
pyasn1              0.2.3
pycparser           2.20
pycryptodome        3.9.8
pyelftools          0.25
Pygments            2.7.1
PyHamcrest          2.0.2
pyparsing           2.4.7
pyrsistent          0.16.0
pyserial            3.4
pytest              6.1.1
python-dateutil     2.8.1
python-dotenv       0.14.0
pyudev              0.22.0
pyusb               1.0.2
PyYAML              4.2b1
requests            2.20.1
semver              2.10.2
setuptools          46.1.3
six                 1.12.0
smmap               3.0.5
soupsieve           2.0
tabulate            0.8.9
toml                0.10.1
tqdm                4.57.0
trollius            2.1.post2
Twisted             20.3.0
txdbus              1.1.2
typing-extensions   3.7.4.3
urllib3             1.24.2
virtualenv          20.4.3
Werkzeug            1.0.1
wheel               0.34.2
wrapt               1.12.1
yattag              1.13.2
zipp                3.1.0
zope.interface      5.2.0
```
#### How is this defect reproduced ? 

Attempt to run above program. Printed results will not match expected. The time will stay 0.

AGlass0fMilk, 2021-03-31:
Need to change the printf statement... the type was wrong. This works:

```
    tr_info("chrono time: %lld", (long long) duration_cast<milliseconds>(system_clock::now().time_since_epoch()).count());
```

closed 2021-03-31 by AGlass0fMilk

kjbracey-arm, 2021-04-01:
>  Is std::chrono::system_clock supposed to work?

No, would be my answer. Tool variance means the `std` ones aren't always available. If it is working, I'm vaguely wondering why.

I wrote some stuff about it here:

https://github.com/ARMmbed/mbed-os/blob/master/platform/docs/clocks/clocks.md#comparison-to-standard-c-chrono-clocks