ARMmbed/mbed-os #14494: Add MBED_TEST_MODE macro when compiling to run Greentea tests

MarceloSalazar, 2021-04-01:
**Is your feature request related to a problem? Please describe.**

In some cases, we need to conditionally include code depending on whether Greentea is used or not ([example](https://github.com/ARMmbed/mbed-os/issues/14492)).
This is also useful to skip the main application without having to delete/rename main.cpp ([example](https://github.com/ARMmbed/mbed-os-example-pelion/blob/master/main.cpp#L18)).

We have the ability to do this in Mbed CLI 1 as it includes the `MBED_TEST_MODE` macro.
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2880

**Describe the solution you'd like**

Define the `MBED_TEST_MODE` macro when compiling Greentea tests (equivalent of `mbed test`).

**Describe alternatives you've considered**

NA - I think we need this to avoid backwards compatibility issues in Mbed OS.

0xc0170, 2021-04-06:
https://os.mbed.com/docs/mbed-os/v6.9/debug-test/greentea-for-testing-applications.html -  I did not know we defined a test building macro.

We will triage this.

LDong-Arm, 2021-04-14:
> This is also useful to skip the main application without having to delete/rename main.cpp

An advantage of CMake is we can control which files to compile or not based on certain conditions (e.g. what CMake target, i.e. build product, we want), so using a macro to "comment out" a whole application file shouldn't be necessary.

Let me check other use cases of this macro though. In my personal opinion, well-structured build logic is preferred.

LDong-Arm, 2021-04-16:
> `mbed test` is a thin wrapper over `ctest`. The tools won't know if they are building greentea or not.

Could you please show me where the thin wrapper is? I thought CTest was only used for [unit tests](https://github.com/ARMmbed/mbed-os/blob/9738b27c7df897c29e9769911d6794ba3e5b3f19/UNITTESTS/CMakeLists.txt#L69), and Greentea tests are just normal Mbed apps that uses [greentea-client](https://github.com/ARMmbed/mbed-os/tree/master/features/frameworks/greentea-client) so it can talk to the host tester through serial. So I'm not sure `BUILD_TESTING` works for Greentea?

0xc0170, 2021-04-26:
Closed? PR is still opened: https://github.com/ARMmbed/mbed-os/pull/14538

0xc0170, 2021-04-26:
I know why. I closed the issue in the backlog as I though it was duplicated but it was attached to this one. I'll reopen

PR #14538 (2021-05-11): Cmake: Add MBED_TEST_MODE macro

### Summary of changes 

Fixes #14494

MBED_TEST_MODE was used in CLI1 to prevent multiple applications from
being built. Add the macro to provide backward compatibility, along with a test 
application.

#### Impact of changes 
None

#### Migration actions required 
None

### Documentation 

CLI1 Reference: https://os.mbed.com/docs/mbed-os/v6.9/debug-test/greentea-for-testing-applications.html

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix /  [... cut]
closed 2021-05-11 by 0xc0170