ARMmbed/mbed-os #14500: STM32H743: Using HSI clock source breaks serial

multiplemonomials, 2021-04-02:
### Description of defect

We've been testing a custom board using the H743ZI, and were having some trouble with the external clock on our board.  So, we tried enabling internal clock (`"target.clock_source": "USE_PLL_HSI"`).  Surprisingly, this seems to break the serial port and prevent it from printing legible characters.  I'm not quite sure what's up with this, maybe SystemCoreClock is getting configured wrong or something's going wrong with the baudrate calculation?

Oh also, one of my boards doesn't even boot when using the default HSI settings -- I think the HSI has some error and the clock is going slightly above 480MHz so the chip is crashing.  Adding an extra divider of 2 using the PLL makes the board boot, but it still isn't able to use the serial port.

#### Target(s) affected by this defect ?

NUCLEO_H743ZI2

#### Toolchain(s) (name and version) displaying this defect ?

GNU 9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cmake 1.6.0

#### How is this defect reproduced ? 

Add `"target.clock_source": "USE_PLL_HSI"` in mbed_app.json.  Then try to run any program that uses the serial port.

jeromecoutant, 2021-04-06:
> Add `"target.clock_source": "USE_PLL_HSI"` in mbed_app.json. Then try to run any program that uses the serial port.

Hi
I have used NUCLEO_H743ZI2 with "USE_PLL_HSI", and greentea tests (using then serial port),
I couldn't reproduce the issue...

multiplemonomials, 2021-04-06:
Huh, I checked again and I still get the issue with GCC and Mbed 6.5, but don't get it with ARMClang and Mbed 6.9.  Must have gotten fixed already!

closed 2021-04-06 by multiplemonomials

multiplemonomials, 2021-04-06:
Huh, I spent a good chunk of today drilling down into the issues between the code that worked and the code that didn't.  Weirdly enough, it seemed to be a serial code problem.  The code that didn't work created a BufferedSerial on USBTX and USBRX, and the code that did just used Mbed's printf().  Since Mbed always creates an internal serial port object on USBTX and USBRX, it looks like there might have been some kind of undefined behavior / race condition from having two serial objects active at a time that caused it to print garbage.  This behavior only showed up when the HSI oscillator was enabled, but went away when I removed the second BufferedSerial object and used printf() exclusively.