ARMmbed/mbed-os #14501: STM32L0 :  flashiap component

atoy40, 2021-04-03:
Hello,

I was using the old nvstore with mbed5 and a Nucleo L037RZ board. I tried to migrate to the new kvstore api, but it seems to require the FLASHIAP component to work, and it is not implemented on the STM32L0. Is it an hardware limitation or a planned feature ?
Thanks
Anthony.

jeromecoutant, 2021-04-06:
Hi
Maybe you could just add "components_add": ["FLASHIAP"],
in the targets.json an propose a pull request ?

| target              | platform_name | test suite                                                        | result | elapsed_time (sec) | copy_method |
|---------------------|---------------|-------------------------------------------------------------------|--------|--------------------|-------------|
| NUCLEO_L073RZ-ARMC6 | NUCLEO_L073RZ | drivers-tests-tests-mbed_drivers-flashiap                         | OK     | 38.09              | default     |

jeromecoutant, 2021-04-06:
Warning: drivers-tests-tests-mbed_drivers-flashiap seems OK, but not storage-kvstore-* tests...

0xc0170, 2021-04-07:
>Warning: drivers-tests-tests-mbed_drivers-flashiap seems OK, but not storage-kvstore-* tests...

 is this related to enabling flashiap or not?

jeromecoutant, 2021-04-07:
> is this related to enabling flashiap or not?

Yes, enabling FLASHIAP makes mbed_drivers-flashiap tests OK, but not storage-kvstore-* tests

closed 2022-07-22 by 0xc0170