ARMmbed/mbed-os #14503: Mail and MemoryPool template classes do not work with classes as template type

JojoS62, 2021-04-06:
### Description of defect

The MemoryPool is a template class that works like in the provided example. But when the type is changed from a structure to a class, the code compiles but it will most likely fail.
The reason is, that the constructor of a class or struct is not called when the MemoryPool is initialized. When the class T itself allocates memory in its constructor, it will have an uninitialized pointer. Calling the constructor after getting the memory block is no option, it will fail in interrupt context.

I have created a fix that will call the constructors when T is a non trivial type and I will prepare a PR for this.

#### Target(s) affected by this defect ?

all, generic RTOS component

#### Toolchain(s) (name and version) displaying this defect ?

gcc 9.x

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli

#### How is this defect reproduced ? 

using e.g. MIDIMessage as template type will crash when used with mboxMIDIMessages.try_alloc():

```
Mail<MIDIMessage, 64> mboxMIDIMessages;
```

0xc0170, 2021-04-07:
> I have created a fix that will call the constructors when T is a non trivial type and I will prepare a PR for this.

Thanks!

JojoS62, 2021-04-12:
closed, explained in #14509

closed 2021-04-12 by JojoS62