ARMmbed/mbed-os #14508: STM32G474 LPUART freezes program

pilotak, 2021-04-07:
### Description of defect

When setting up LPUART1 as debug console it cause program to stop. Please see example bellow. If you actually remove the `printf` command (which means there is nothing to print out) it works fine (LED is blinking)

This problem is not only when console is LPUART, but also when is used as regular serial port ie.`BufferedSerial` (of course console is redirected back UART2). In this case the progam is freezed but unfortunatelly no error is reported via console.

#### Target(s) affected by this defect ?
STM32G474

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

84e5f1e03eed2227d91937c38ab862297ee9c2d6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 
```json
{
  "target_overrides": {
    "*": {
      "platform.error-all-threads-info": true,
      "platform.error-filename-capture-enabled": true,
      "platform.fatal-error-auto-reboot-enabled": true,
      "platform.stdio-convert-newlines": true,
      "platform.stdio-baud-rate": 115200,
      "platform.callback-nontrivial": true,
      "drivers.uart-serial-rxbuf-size": 2048,
      "drivers.uart-serial-txbuf-size": 256,
      "target.printf_lib": "std"
    },
    "NUCLEO_G474RE": {
      "target.stdio_uart_tx": "PC_1",
      "target.stdio_uart_rx": "PC_0",
      "target.clock_source": "USE_PLL_HSE_XTAL|USE_PLL_HSI",
      "target.lpuart_clock_source": "USE_LPUART_CLK_PCLK1"
    }
  }
}

```

```cpp
DigitalOut led(PB_14, 1);

int main() {
    while (1) {
        printf("test\n");
        led = !led;
        ThisThread::sleep_for(500ms);
    }
}
```

jeromecoutant, 2021-04-07:
Hi

>       "target.lpuart_clock_source": "USE_LPUART_CLK_PCLK1"

Could you try ?
"lpuart_clock_source": "USE_LPUART_CLK_HSI"

pilotak, 2021-04-07:
@jeromecoutant didn't help, same result...

pilotak, 2021-04-22:
Any update on this? a really need to push it into production and this is a real blocker. I would appeciate if you could look into this, i will of course test.

jeromecoutant, 2021-04-22:
Hi
Could you try https://github.com/jeromecoutant/mbed/commit/f2e153cf741679db0f5e6bf8e39d5d9a9e63b9bd
Thx

pilotak, 2021-04-22:
perfect, thank you very much

PR #14581 (2021-04-28): STM32G4 : serial issue with LP_UART1
### Summary of changes 

Fixes #14508

There was an issue with the handler and IRQ table size.

#### Impact of changes 

We can now use LP_UART1

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type 

    [x] Patch update (Bug fix / Target update / Docs update / Test update / Refactor)
    [] Feature update (New feature / Functionality change / New API)
    [] Major update (Breaking change E.g. R
closed 2021-04-28 by 0xc0170